linuxwacom/wacom-hid-descriptors #1: Lenovo Thinkpad 10

cbm755, 2016-06-16:
The Lenovo Thinkpad 10 tablet seems to have the same Wacom + Atmel combination as the
`Lenovo ThinkPad Helix 2 (version A)`

1. Currently the pen and eraser work
2. Althought possibly without pressure information.
3. Gnome 3 displays the "unknown tablet connected" with each login.
4. I'm running Fedora 24 with `libwacom-0.20`.

(I'm not sure if this information is useful, or indeed exactly what the `wacom-hid-descriptors` project is for, but I found a link to here and thought this information might help.)

`cat /proc/bus/input/devices` gives:
````
---- 8< ----

I: Bus=0018 Vendor=056a Product=0114 Version=0100
N: Name="Wacom HID 114 Pen"
P: Phys=i2c-WCOM0008:00
S: Sysfs=/devices/platform/80860F41:02/i2c-10/i2c-WCOM0008:00/0018:056A:0114.0001/input/input3
U: Uniq=
H: Handlers=mouse0 event3 
B: PROP=2
B: EV=b
B: KEY=c03 0 0 0 0 0
B: ABS=1000003

---- 8< ----

I: Bus=0018 Vendor=03eb Product=8c15 Version=0100
N: Name="ATML1000:00 03EB:8C15"
P: Phys=i2c-ATML1000:00
S: Sysfs=/devices/platform/80860F41:05/i2c-13/i2c-ATML1000:00/0018:03EB:8C15.0003/input/input8
U: Uniq=
H: Handlers=mouse1 event6 
B: PROP=2
B: EV=b
B: KEY=400 0 0 0 0 0
B: ABS=260800000000003

---- 8< ----
````

jigpu, 2016-06-16:
Thanks for the information :) If you don't mind, could you provide me with the specific model number of your tablet and the output of the following commands? Also, any additional information you might know would be appreciated as well (e.g. if you know if the pen uses EMR [battery-free] or AES [requires batteries or periodic charging] technology).

    $ hexdump -C '/sys/devices/platform/80860F41:02/i2c-10/i2c-WCOM0008:00/0018:056A:0114.0001/input/input3/../../report_descriptor'
    $ hexdump -C '/sys/devices/platform/80860F41:05/i2c-13/i2c-ATML1000:00/0018:03EB:8C15.0003/input/input8/../../report_descriptor'

I really should add a README, but this project basically just exists to be a repository of information about Wacom sensors seen in the wild. The IDs and HID descriptors in particular are useful for writing new libwacom ".tablet" definitions (to get rid of those "unknown tablet connected" messages :)), backporting support to pre-3.17 Linux kernels, and validating assumptions about devices that we developers don't have in our possession. Someday I'd like to also store `hid-replay` traces of devices here for automated testing but that's a future project...

cbm755, 2016-06-16:
Model number: not sure precisely but from dmidecode:
````
System Information
	Manufacturer: LENOVO
	Product Name: 20C1CTO1WW
	Version: ThinkPad 10
	Serial Number: [ ---- 8< ----]
	UUID: [ ---- 8< ----]
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_20C1_BU_Think_FM_ThinkPad 10
	Family: ThinkPad 10
````
Consumer-wording wise its something like "Thinkpad 10, 64-bit version".

I believe the pen is battery-free: the pens from my regular Thinkpad (x230t) work fine.

I did look at making a `.tablet` file but it wasn't obvious what the bus is (i2c?).

Doing hexdumps next...

cbm755, 2016-06-16:
The numbers had changed a bit across a reboot: the number *after* `0114` should *not* be used for matching....

````
hexdump -C /sys/devices/platform/80860F41\:02/i2c-10/i2c-WCOM0008\:00/0018\:056A\:0114.0003/input/input7/../../report_descriptor
hexdump -C /sys/devices/platform/80860F41\:05/i2c-13/i2c-ATML1000\:00/0018\:03EB\:8C15.0005/input/input10/../../report_descriptor
````

[0018:056a:0114.0003_input7.hex.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/319071/0018.056a.0114.0003_input7.hex.gz)
[0018:03eb:8c151.0005_input10.hex.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/319070/0018.03eb.8c151.0005_input10.hex.gz)

cbm755, 2016-06-16:
Much more detail about the hardward (overall machine, not just the wacom) is here:
http://nf.dpin.de/new-try-thinkpad-tablet-10-1st-gen/

(reasonably confident this is the same hardware).

cbm755, 2016-06-16:
Note Lenovo's website has "2nd gen" "Thinkpad 10" with `x7-Z8700` SoC: this is *not* my hardware---mine is `Z3795`: "1st gen" I guess.

jigpu, 2016-06-16:
Awesome, thank you for all that :) I'll compile everything and push a new folder shortly. In the meantime, I've put together a '.tablet' file that should work for you.
[isdv4-114.tablet.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/319095/isdv4-114.tablet.gz)

cbm755, 2016-06-16:
Great, thanks.  I can confirm that works.

I trust you'll push that file to libwacom?

jigpu, 2016-06-16:
Descriptors and other information pushed in commit 1fd5ec4. @whot usually handles the libwacom patches, so I'll make sure he gets his hands on it.

closed 2016-06-16 by jigpu

cbm755, 2016-06-16:
@jigpu As I noted above, those `0005` and `0003` numbers seem to change upon reboot!   Isn't it a bad idea to record/match/whatever with those?

`0018:056a:0114.hid.xml` a better name?

jigpu, 2016-06-16:
I'm not entirely sure, to be honest. Its easy enough to ignore though, and even if the absolute values don't matter the relative values might actually provide information about the order that the kernel probes a device's constituent pieces if it has multiple interfaces (need to verify that though).