linuxwacom/wacom-hid-descriptors #2: Second hardware ID for Lenovo ThinkPad X1 Yoga panel

rannbaron, 2016-09-18:
The current Wacom hardware ID for this panel is 056a:503e, whereas my machine shows 056a:503f. I believe this results in the panel being unrecognized. Please add it! I'd be glad to provide any information needed as requested. Many thanks!

jigpu, 2016-09-20:
Could you please run [`scripts/sysinfo.sh`](https://github.com/linuxwacom/wacom-hid-descriptors/blob/master/scripts/sysinfo.sh) from this repository and post the generated tarball for review? If you have a kernel older than Linux 3.17, it would explain why you do not have any response at all from the pen or touch since our sister [input-wacom](https://github.com/linuxwacom/input-wacom) repository does not support the 056a:503f.

rannbaron, 2016-09-20:
I'm running kernel 4.7. 

sysinfo file is attached.
[sysinfo.GFqIeaj6nj.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/483335/sysinfo.GFqIeaj6nj.tar.gz)

jigpu, 2016-09-20:
That's weird. The 4.7 kernel should definitely be capable of using your pen and touch sensors since they have valid HID descriptors. Reading your post again more closely, it sounds like the cursor tracks the position of the pen? If it does, its probably an issue with userspace rather than the kernel. Just to be sure though, please try installing the `evemu-tools` package and running `evemu-record > /tmp/pen.log` for me. When the program starts, it should ask you to choose a device -- enter the number for the pen device and press enter. Bring the pen into proximity and touch it to the screen, then press the lower and upper button in order. Finally, remove the pen from proximity, press CTRL+C, and upload the generated `/tmp/pen.log` file for me to review.

rannbaron, 2016-09-20:
[pen.log.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/483791/pen.log.txt)

jigpu, 2016-09-21:
The log indicates that the kernel is sending pressure and button information to userspace, so any issues that you're having must be higher up in the stack. What output do you see when running `xsetwacom list`? If you don't see anything, please run the following command and attach the generated file for review.

    journalctl -b0 _COMM=Xorg _COMM=Xorg.bin _COMM=gdm-x-session | gzip > journal.log.gz

This should let me know if the xf86-input-wacom driver is picking up your device, or if some other driver is taking priority.

closed 2017-06-02 by jigpu