linuxwacom/wacom-hid-descriptors #3: Data for Wacom Graphire USB ET-0405-U

schreiberste, 2017-04-16:
Hi,

I'm not sure, if this is the right project to contact, but I came here while searching for my problem, which is Gnome 3 showing "unknown tablet connected" on each login.

The tablet is an rather old Wacom Graphire USB, Model ET-0405-U.
I attached all output of the sysinfo.sh script.
Unfortunately, github won't let me add all files in one .tgz or .zip, and I also can't add the 0003:056A:0010.0004.hid.bin file.
Just tell me, if you need more information.

Thanks,
Steffen

[host.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/924159/host.txt)
[kernel_drivers.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/924160/kernel_drivers.txt)
[libinput.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/924161/libinput.txt)

[lsusb.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/924162/lsusb.txt)
[lsusb_tree.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/924163/lsusb_tree.txt)
[machine.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/924164/machine.txt)
[packages.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/924165/packages.txt)
[udevadm.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/924166/udevadm.txt)
[udevadm_0003:056A:0010.0004.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/924167/udevadm_0003.056A.0010.0004.txt)
[userspace_drivers.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/924168/userspace_drivers.txt)
[xinput.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/924169/xinput.txt)
[xsetwacom.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/924170/xsetwacom.txt)

schreiberste, 2017-04-16:
I added a new file /usr/share/libwacom/graphire-usb-8x6.tablet with the following content, and the device was recognized by Gnome:

[Device]
# Model ET-0405-U (Grapphire-USB) (ET-0405-U00)
Name=Wacom Graphire-USB 8x6

DeviceMatch=usb:056a:0010

Class=Graphire

Width=8
Height=6

[Features]
Stylus=true
Buttons=0

jigpu, 2017-04-17:
Thanks for both the sysinfo.sh script output and the tablet definition file. I'll work on adding the tablet definition to libwacom so GNOME doesn't display that warning in future releases and update this bug as that process moves along.

jigpu, 2017-04-17:
Also, if you're having trouble uploading 0003:056A:0010.0004.hid.bin, please try using the following command to create a text hex dump:

    $ hexdump -C 0003:056A:0010.0004.hid.bin > /tmp/0003:056A:0010.0004.hid.bin.txt

I can convert it back to its binary form without trouble :)

schreiberste, 2017-04-18:
Thanks for taking care!
Uploaded the hexdump text file.

[udevadm_0003:056A:0010.0004.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/930743/udevadm_0003.056A.0010.0004.txt)

closed 2017-06-02 by jigpu