linuxwacom/wacom-hid-descriptors #6: Dell XPS 13 9365 support

vdanjean, 2017-09-13:
Hi,
I'm not sure it is the right place to report (if not, please tell me where), but here is the result of the sysinfo.sh script on this laptop.
[sysinfo.fQZQP6aOMQ.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/1299450/sysinfo.fQZQP6aOMQ.tar.gz)

  If nor need more info, just tell me. For now, gnome talk me about unsupported tablet at each login.

  Regards,
    Vincent

jigpu, 2017-09-13:
Thanks for the sysinfo output. I've created the necessary patch to have libwacom support your hardware and sent it to the mailinglist for review. I've also added the data to this repository for future reference. I'll update this bug once the libwacom patch is accepted.

jigpu, 2017-09-14:
Patch accepted and expected in libwacom 0.27: linuxwacom/libwacom@d14f423d692ff1f2121cc49bf22eff9844cb38da

closed 2017-09-14 by jigpu