linuxwacom/wacom-hid-descriptors #8: HP Spectre X2 2017 no touch screen

kevinmarvin, 2017-11-20:
Based on the post I found here: 
[The HP Spectre x2 (2017 Kaby Lake) and Linux: a quick summary](https://www.reddit.com/r/linuxhardware/comments/78dv9f/the_hp_spectre_x2_2017_kaby_lake_and_linux_a/)

I have the device mentioned in the post and can confirm that the drivers do not support this touch panel.
I have followed the advice of one of the posters and run the sysinfo.sh script.  Attached is the result.
[sysinfo.qpWPtdJ23p.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/1488408/sysinfo.qpWPtdJ23p.tar.gz)

Any help is appreciated.

kevinmarvin, 2017-11-20:
Here's the same file, but with the script run under sudo :( 

[sysinfo.bMG7czIv8B.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/1488412/sysinfo.bMG7czIv8B.tar.gz)

jigpu, 2017-11-20:
Thanks for the output from that device. For whatever reason, it looks like the sysinfo.sh script wasn't able to find any indication of Wacom hardware on this device. It certainly looks like this this model uses a Wacom pen, so I have to wonder what's going on... Would you be able to run `sudo journalctl -b0 > /tmp/journal.log` after a fresh reboot and attach the generated file? Hopefully it has some hint (e.g. driver failures, etc). I also notice that the lsusb_tree.txt output contains a "full speed" HID device with two interfaces (similar to how some Wacom tablets appear to the system) -- could you run both `lsusb` and `lsusb -t` so I could get an idea of what this might be?

kevinmarvin, 2017-11-20:
Here are the files you requested, made as quickly after a fresh boot as possible:

[step2.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/1489344/step2.tar.gz)

jigpu, 2017-12-11:
Sorry for the delayed response... I don't see any hint of Wacom hardware in these new logs. The Full-speed device with two interfaces looks to be something different. The device labels itself an Elan Microelectronics "USBSmartKeyboard".

I found [a post on HP's forum about someone having issues with the pen](https://h30434.www3.hp.com/t5/Notebook-Video-Display-and-Touch/HP-Spectre-X2-Consumer-Version-Stylus-Not-Working/td-p/6329675), and their screenshots seem to indicate that this device might be using "Intel Precise Touch", despite the pen appearing to be one of Wacom's AES-type. I see a PCI ID in your logs which match the screenshot (8086:9d3e).

I don't know anything about this "Intel Precise Touch" technology... Perhaps its compatible with Wacom's pens, but if the controller is non-Wacom then our drivers aren't necessarily going to be able to make it work. You might see if you can find out any more about this device. You could also see if there's anything you can learn from sysfs (see if there's any hits for `find /sys/devices -iname "*9d3e*"` and then go exploring from there).

kevinmarvin, 2017-12-12:
Jason:  Thanks for the feedback, and for the lead.  I think I've found a solution here : [dtrip/linux-ipts](https://github.com/dtrip/linux-ipts)  -  It appears that this Intel Precise Touch system is also used in the Surface Pro 4.  I'll also update the reddit thread with this (if in fact this hand-compiled kernel solves the issue).

Thanks again for all your help!

jigpu, 2017-12-19:
Info pushed to repository as commit 0e0094033d645b158183a72dec960210187aba30

jigpu, 2018-02-07:
It looks like an Elan device was being used for the HP Envy x360 in issue #12. I wonder if its true for this device too (assuming you didn't have any luck with Intel Precise Touch). Could you try running the following command?

~~~
$ find /sys/devices -iname "*04f3*" -or -iname "*0732*" -or -iname "*elan*"
~~~

matsurihime, 2018-12-02:
> ` $ find /sys/devices -iname "*04f3*" -or -iname "*0732*" -or -iname "*elan*" `
I tried this, and got
```
/sys/devices/pci:0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:04F3:0757.0001
/sys/devices/pci:0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:04F3:0757.0002
```

jigpu, 2018-12-03:
Thanks @matsurihime -- that's definitely an Elan device. If you run the [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) I can add the data to this repository. The script has been updated quite a bit since kevinmarvin ran it and it should be able to read more useful information about this Elan sensor.

Note that our Elan sensors don't use our drivers (at least, not by default) but the information gathered by the script may still be useful to other developers (e.g. those at Digimend, if they support Elan).

matsurihime, 2018-12-03:
Thanks for your reply, @jigpu . 
Here is my result generated by latest sysinfo.sh script.
[sysinfo.Ft79325KH0.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/2640256/sysinfo.Ft79325KH0.tar.gz)

jigpu, 2018-12-04:
Thanks -- added to the repository and closed again.

closed 2018-12-04 by jigpu