linuxwacom/wacom-hid-descriptors #12: HP Envy x360 (Ryzen 2500U APU) touchscreen support

jhendersonHDF, 2017-12-19:
It seems that the touchscreen functionality does not currently work for this newly-released laptop. I have built linux 4.15-rc4 with all available touchscreen-related drivers I could find in the kernel config but none seem to work. Attached is the result of running the sysinfo script on Slackware64-current. Though Slackware is a rarity, functionality also did not work on Ubuntu 17.10 and I did not see any mention of support (of course this is a very new hardware configuration).

[sysinfo.XXXXEdoyNI.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/1570354/sysinfo.XXXXEdoyNI.tar.gz)

DrewTechs, 2017-12-19:
Wow, you beat me to it by a short margin, I been trying to figure this one out myself.

jigpu, 2017-12-19:
Thanks for the sysinfo output. It doesn't look like it was able to detect any Wacom hardware present...

I wonder if, like the HP Spectre X2 2017 in bug #8, this particular tablet PC has an "Intel Precise Touch" controller. I don't see a driver for that listed on HP's support page for this device, but I might be looking at the wrong one (there are sever Envy X360 PCs listed).

Could either of you please attach the output of `dmesg` for review? Also, if either of you still has Windows installed on a partition, it would be helpful to see if you can find any pen devices in the Device Manager. If you do, double-click on the device, go to the "Details" tab, select the "Hardware IDs" property, and take a screenshot.

jigpu, 2017-12-19:
Info (for both this and issue #13) pushed to repository as commit 369eceae8c5c76a9fda67c0442b28557ff7b6776

DrewTechs, 2017-12-19:
Here is the output for dmesg on my system:

[dmesg_output.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/1573138/dmesg_output.txt)

jigpu, 2017-12-19:
Thanks :) No trace of devices with Intel's 8086 VID in the dmesg output, so its probably not some kind of Intel Precise Touch thing. I did a little more digging around the HP website, and if I go to purchase this tablet and choose "Details" for the pen accessory, it brings up a [page with the following text](http://store.hp.com/us/en/pdp/hp-pen):

> Reinvent the way you express yourself with the power of HP pen. Write, draw or compute with precision thanks to a natural, true-to-life feel. With our enhanced N-trig technology, your productivity is sure to get an upgrade.

The Spectre X2 2017 links to the same pen, which makes me suspect that these actually use N-trig sensors instead of Wacom sensors. The [HP Active Pen with App Launch](http://www8.hp.com/us/en/products/oas/product-detail.html?oid=11262074) on the other hand is definitely a Wacom pen but it says that it is only compatible with the "HP Elite x2 1012 G1; HP Sprout Pro G2; HP EliteBook x360 1030 G2; HP Elite x2 1012 G1; HP Elite x2 1012 G2".

Please try running the following:

    $ find /sys/devices -iname "*0531*" -or -iname "*056A*" -or -iname "*2D1F*" -or -iname "*WACf*" -or -iname "FUJ*" -or -iname "*1B96*" -or -iname "*NTRG*" -or -iname "*045E*" -or -iname "*MSHW*"

I think that should find Ntrig/Microsoft devices in addition to Wacom ones...

jhendersonHDF, 2017-12-19:
Unfortunately nothing is found with the command. I'm still trying to poke around the devices and see if anything relevant is found under input devices, but no luck so far.

DrewTechs, 2017-12-20:
Nothing was found with this command sadly.

jhendersonHDF, 2017-12-20:
Looking around at other forums on the issue, it seems like the laptop has an ELAN0732 I2C touchscreen controller and looking in the dmesg log occasionally shows a few

[   39.324867] i2c_hid i2c-ELAN0732:00: failed to reset device.
[   40.332682] i2c_hid i2c-ELAN0732:00: can't add hid device: -61

so it seems the x360 has the same problem across models.

DrewTechs, 2017-12-22:
I noticed something when my computer booted up and according to this post, this may have something to do with the pen and touchscreen input not working right.

https://askubuntu.com/questions/848530/hp-envy-x360-15-6-touchscreen-will-not-work

@jhendersonHDF Have you noticed when your computer boots up that an AMD gpio error pops up? I suspect that this may have to do with the touchscreen and pen drivers not working like they should be.

If this is the case, then we may need to tell Kernel developers and/or AMD's software developers about this. Fixing this is a bit beyond my scope right now.

jhendersonHDF, 2017-12-22:
I have indeed noticed this popping up as an error in my logs from time to time, but it doesn't seem to always do so; haven't figured that out yet. I think I remember seeing one solution that just disabled all GPIO support in the kernel when building, causing the kernel to fall back to a different method for the touchscreens in the issue in particular, but I don't know if this is the right solution.

DrewTechs, 2017-12-22:
I am not certain but I was thinking that the proper GPIO support is needed for the touchscreen and pen to work.

jigpu, 2017-12-23:
GPIO support may or may not make a difference for the device. Its highly unlikely the kernel has to use GPIO "bit banging" to communicate with the hardware (I don't believe I've ever seen Wacom hardware connected through anything but a real controller) and though GPIO could be used to power the hardware on and off, in my experience its usually something else.

DrewTechs, 2017-12-24:
I ran libwacom-list-local-devices and this is the output I got:

[libwacom_list_local_devices.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/1584620/libwacom_list_local_devices.txt)

jigpu, 2017-12-24:
The "serial:0000:0000" device that it lists is kind of a catch-all that appears any time it can't specifically detect a device (because you can't detect the kind hardware that is connected to a serial port). My plain old laptop which definitely doesn't have any kind of pen or touch support produces the same output when running libwacom-list-local-devices.

DrewTechs, 2017-12-24:
That's odd then, because I know this laptop has pen support.

Benevolent, 2018-02-06:
Hello I have recently recieved this machine and would like to contribute more information. I am about to reboot to windows and look at what it reports for touchscreen drivers but in the meantime heres my tarball.

[sysinfo.a6Vkti9iAB.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/1697975/sysinfo.a6Vkti9iAB.tar.gz)

Benevolent, 2018-02-06:
Windows reports this under devices as "HID-compliant touchscreen"

I am able to pull some configuration data relating to this device from event viewer:
DeviceInstanceId | HID\ELAN0732&Col01\4&2e44c06e&0&0000
DriverName | input.inf

Let me know if there is any other information I can get you.

jigpu, 2018-02-06:
@Benevolent could you try running the following command from within Linux?

`$ find /sys/devices -iname "*04f3*" -or -iname "*0732*"`

Benevolent, 2018-02-06:
The script returns the following output:

`
/sys/devices/platform/AMDI0010:00/i2c-0/i2c-ELAN0732:00
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/AMDI0010:00/ELAN0732:00
`

jigpu, 2018-02-06:
Aha! So the kernel can at least see it. Next question is why the hid-generic driver doesn't bind to it. Do you see the earlier-mentioned "failed to reset device" and "can't add hid device: -61" messages in dmesg?

Benevolent, 2018-02-07:
Not with 4.15. I am attatching my DMESG log. Please note the swath of ACPI Errors at around 3.495 and on.

[dmesg.log](https://github.com/linuxwacom/wacom-hid-descriptors/files/1701466/dmesg.log)default

Benevolent, 2018-02-07:
It is probably worth mentioning that I am booting the following kernel https://aur.archlinux.org/packages/linux-ryzen-git/

jigpu, 2018-02-07:
Thanks for checking. ACPI errors [seem to be unrelated](https://bugzilla.kernel.org/show_bug.cgi?id=194545), and I don't see any other interesting messages about this device in dmesg.

Since we now have at least a clue of what this device is, I think its best to move further discussion to the Kernel bugzilla. Please [file a bug](https://bugzilla.kernel.org/enter_bug.cgi?product=Drivers) about the touchscreen/pen not working, with any relevant system information and data gathered from Windows/Linux.

@jhendersonHDF and @DrewTechs you'll want to reply to the kernel bug that gets created to ensure you're notified of updates.

I'll be temporarily closing this issue until the kernel is in a state that we can get useful sysinfo output.

closed 2018-02-07 by jigpu

Benevolent, 2018-02-07:
[Figured I would leave a link here](https://bugzilla.kernel.org/show_bug.cgi?id=198715) just in case someone came by with the same issue.

brauliobo, 2018-04-25:
Weirdly, `xsetwacom --list devices` still returns nothing

jigpu, 2018-04-25:
As an Elan sensor, it's probably not going to be handled by the xf86-input-wacom driver (and thus listed in xsetwacom) unless you specifically configure your system that way.

Id still be interested in the sysinfo.sh output from a working system for future reference, however.

brauliobo, 2018-04-25:
@jigpu never used sysinfo.sh? how to download and use it at archlinux?

brauliobo, 2018-04-25:
for now, I post related dmesg messages:
```
~/g/scripts ❯❯❯ dmesg | grep ELAN                                                                                                                                          master ✱ ◼
[    2.978216] i2c_hid i2c-ELAN0732:00: i2c-ELAN0732:00 supply vdd not found, using dummy regulator
[    3.244303] input: ELAN0732:00 04F3:2537 as /devices/platform/AMDI0010:00/i2c-0/i2c-ELAN0732:00/0018:04F3:2537.0002/input/input11
[    3.244539] hid-generic 0018:04F3:2537.0002: input,hidraw1: I2C HID v1.00 Device [ELAN0732:00 04F3:2537] on i2c-ELAN0732:00
[    3.424630] input: ELAN0732:00 04F3:2537 as /devices/platform/AMDI0010:00/i2c-0/i2c-ELAN0732:00/0018:04F3:2537.0002/input/input14
[    3.455991] input: ELAN0732:00 04F3:2537 Pen as /devices/platform/AMDI0010:00/i2c-0/i2c-ELAN0732:00/0018:04F3:2537.0002/input/input17
[    3.456730] hid-multitouch 0018:04F3:2537.0002: input,hidraw0: I2C HID v1.00 Device [ELAN0732:00 04F3:2537] on i2c-ELAN0732:00
```

jigpu, 2018-04-25:
@brauliobo the script can be found at https://github.com/linuxwacom/wacom-hid-descriptors/blob/master/scripts/sysinfo.sh

Just run it and attach the tarball it creates to this issue.

brauliobo, 2018-04-25:
Nice, there it is

[sysinfo.AYPzYTeFcO.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/1949090/sysinfo.AYPzYTeFcO.tar.gz)

dan-and, 2018-05-11:
Just want to mention that the touchscreen works fine on the Envy X360 using the patch in following bugreport: https://bugzilla.kernel.org/show_bug.cgi?id=198715

brauliobo, 2018-05-11:
@dan-and that patch/bugreport was mentioned before

robertdmunn, 2018-06-24:
I am running Ubuntu 18.4 on this system - Envy x360 Ryzen. I installed kernel 4.17.0 to get most functionality working. I have not patched the kernel yet for the touchscreen functionality. I can help test other fixes, etc. if you all need additional help.

sashamishchenko, 2018-07-06:
@robertdmunn 
Hi!
Please, answer for my questions.
Are you had fully working and stable Ubuntu 18.04 on HP Envy x360 Ryzen? Are you has 15" notebook or 13"?
Are Ubuntu 18.04 stability depends from 4.17 kernel? Are it works with issues from box?
Are all ACPI keys works? Are LID-key work if run it in notepad mode? Can it automatically disable keys and touchpad in this mode?
How long it can work from battery?
May be you can post another your first impressions.
Thank you in advance.

robertdmunn, 2018-07-07:
15" HP Envy x360 Ryzen 2500U, 32 GB RAM. I don't normally depend on my battery because I mostly work at a desk, but I think I could go 5 hours, maybe longer depending on settings ,what I am doing.

It is mostly stable but experiences some random freezes in both Ubuntu and Windows 10. I haven't pinned down exactly what is going on there. In Ubuntu, I was experiencing system freezes using a Windows 2008 guest VM in VirtualBox. I coverted the VM to KVM and that seems to have fixed it. Other freezes *seem* to be related to graphics drivers/Chrome issues. It freezes in Ubuntu on a WebGL test in Chrome, for instance.

I had to install the 4.17 kernel to fix some issues, though most stuff worked off the bat after 18.04 install. Right now I'm running 4.17.4-041704-generic, hoping 4.18 is released soon with graphics driver updates. 

The touchscreen does not work because of an ACPI issue, I'm waiting for either a kernel update or ACPI fix from HP to fix that problem.

I don't know about tablet mode, I never use it that way. If you have very specific questions about it I might be able to test things, but it just isn't part of my normal workflow.

The laptop came with a relatively slow SSD (WD Blue), so I upgraded to a Samsung 970 Pro. I bought a second WD Blue SSD and a mini RAID device to plug the SSD pair into the 2.5" drive space and improve speed on the second drive at minimal cost. Upgrade process is quite easy for anyone with some experience with computer hardware maintenance. 

The things I don't like about the hardware:
1. The numeric keypad. I don't do enough number data entry to use it regularly, and it shifts the rest of the keyboard left, forcing me to shift my wrists left on the keyboard compared to my MacBook Pro. It take an hour or so when switching between the machines for me to adjust to the difference. 
2. The power button location. The button is on the left side of the machine next to one of the USB ports. I find myself inadvertently hitting it on occasion when I pick up/grab the laptop. 
3. The feel of the trackpad. The surface material falls short of the smooth glide feel of the MB Pro trackpad, and the tactile feel of clicking is not quite as nice. Add the shift of where my hands sit on the machine and I find myself clicking the wrong part of the trackpad on occasion. 

The hardware issues are not really a big deal to me since I normally work with an external mouse/keyboard, but they are worth noting. 

Things I like about the hardware:

1. Backlit keyboard. I wish my MB Pro had this feature. 
2. 32 GB RAM. Again, I wish my MB Pro had this feature. This is the main reason I'm not using my MB Pro for development work now. 16 GB is just not enough anymore.
3. Ryzen 2500U. Very nice, capable CPU. Intel has been needing competition in this space for several years. 

Overall I am very happy with this system. Even with the upgraded SSDs, it's still half the price of what would be a comparable MB Pro (once Apple releases a system with 32 GB RAM). 

I have been trying to move from WIndows 10 to Ubuntu 18.04 as my daily OS for development, but there are a couple of issues holding me back. I use the Slack desktop client, and it is in beta on Linux and not everything works. Moving my Windows VM to KVM may be the last step for development work, though I would like to see whether the 4.18 kernel updates ( and perhaps 4.19 ) resolve the graphics driver stability issues. Like I said, I've seen some freezes in Windows, too, so that may just be how it is with the newness of the hardware components. 

Also, it's worth noting that Windows 10 is pretty good and mostly works very well with this hardware. There are a lot of improvements compared to Windows 7, though some of the annoying limitations remain.

brauliobo, 2018-07-08:
@robertdmunn for freezes, use `idle=nomwait` (from https://community.amd.com/thread/224000)

robertdmunn, 2018-07-09:
@brauliobo, added last night, no crashes since then. :-)

Also, I compiled and installed the latest (18.20) AMD Vega drivers from source:

https://support.amd.com/en-us/download/linux

I've installed a few Gnome shell extensions:
- gno-menu
- todo.txt
- netspeed
- openweather 
- clipboard indicator

I've also install the ubuntu-restricted-extras.

I'm hesitant to do too much more until I run the system for a few days and verify stability,

sashamishchenko, 2018-07-09:
@robertdmunn 
Thanks! This is very nice and clear review!
Worst that touch screen still does not work. I believe that this bug will be resolved in nearest future.
I was amazed that SSD in Envy are slow. I believe that it has NVMe ssd...
I am looking for Envy x360 13" version. But, I think, both has similar issues. May be 13" version has a bit smaller TDP on CPU.
Now only HP Envy x360, Dell 2-in-1 7000 and Lenovo 720s has 13" versions. Last is smaller and thinner than all of them. But it has lack - bad cooling system, which decrease CPU power. Also it has single channel RAM. ((( Dell has good hardware realization, but it is not thin - 20mm. HP Envy can be as a compromise for me, but it has not resolved disadvantages too...

robertdmunn, 2018-11-05:
I wanted to provide an update on my experiences with this machine after a few months of solid use.

The underside gets slightly warm (not hot) with my 3-SSD solution (see above). I'm going to see if I can add a heat-spreader to the M.2 SSD. I don't know what I can do about the other ones since they are in a 2.5" disk housing, but it's worth a look.

The touchscreen ACPI issue in Linux is still open but now has attention from the Linux kernel ACPI maintainers, so I expect it will get fixed sometime soon. Apparently the bug affects older AMD-based laptops, as well.

I upgraded to Ubuntu 18.10, which runs the latest kernel and Gnome - those are worthwhile updates. 

I still experience the occasional problem/freeze in Ubuntu if I close the lid- rather than going to sleep it becomes non-responsive. Not a big deal for me as that is not something I normally do with my laptop, but it would be unacceptable if I needed that feature to work on a daily basis. I have not reported it or done any debugging on it.

I haven't been using my WIndows VM in Ubuntu, as Windows Server 2008 reports that I am running unsupported hardware. I may get around to installing a Windows 10 VM, but having dual boot with Windows on a separate disk makes it an extra, not a requirement.

Overall, I've been very happy working on it. I work about 50% of the time in Windows and 50% in Ubuntu. Windows 10 has come a long way, but I like Ubuntu.

luyatshimbalanga, 2018-11-24:
[sysinfo.67UCNCKGdn.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/2612159/sysinfo.67UCNCKGdn.tar.gz)

Added information of HP ENVY x360 Convertible 15-cp0000 with fixed both touchscreen and stylus support. Gnome Shell detected them but cannot get more information as a non-wacom device.

tabuvudu, 2018-11-27:
HP envy x360 13-ag0018au with Ryzen 7 here, the touchscreen issue is resolved with BIOS F.21.Rev.A and standard fedora kernel-4.19.3-300.fc29.x86_64

jigpu, 2018-11-27:
Thanks @luyatshimbalanga  -- I've added your sysinfo to the repository as well.

luyatshimbalanga, 2018-12-01:
No problem @jigpu

Eusebius1920, 2018-12-20:
FYI, the problems resurfaced:
https://bugzilla.kernel.org/show_bug.cgi?id=201817

Touchscreen not wokring atm for me (HP Envy x360 13-ag0004ng) on kernel 4.19.10 (archlinux)

brauliobo, 2018-12-21:
it is now working with vanilla kernel 4.9.10 and BIOS F.19 :smile:

tacchinotacchi, 2018-12-26:
The issue came back for me on BIOS f.31 and kernel 4.20.
I sometimes suspect OEMs are getting off of our suffering on these threads

jgaynon, 2018-12-29:
Can confirm that BIOS update F.31 broke the touchscreen again with kernel 4.20. HP in their infinite wisdom don't allow BIOS downgrades either. Ridiculous.

epsilon-0, 2019-01-17:
This utter madness of HP is ripping out my heart with Kernel 4.20.2 and Bios F.31

May the programmers of HP feel my wrath through the curse I lay down upon them for breaking this.

MrTomRod, 2019-01-17:
I have a HP Spectre x360 – 15-df0709nz with a ELAN2514 touchscreen. The latest BIOS update broke my touchscreen as well, but I found a workaround: Enable **legacy boot support** in the BIOS. Maybe it works for you guys, too...

epsilon-0, 2019-01-17:
> 
> 
> I have a HP Spectre x360 – 15-df0709nz with a ELAN2514 touchscreen. The latest BIOS update broke my touchscreen as well, but I found a workaround: Enable **legacy boot support** in the BIOS. Maybe it works for you guys, too...

Do you wanna tell if you did anything more?
I tried enabling legacy boot and nothing happens....

Eusebius1920, 2019-01-17:
I just wanted to add, that for me everything still works (UEFI **not** legacy boot):

BIOS: F.31
Notebook: HP ENVY x360 13-ag0004ng
Kernels: 4.20.0-arch1, 4.20.3-arch1 (which is almost the mainline kernel)

It just doesn't work if I booted windows right before linux (as some suggest in the redhat bugtracker that this might be a problem:
https://bugzilla.kernel.org/show_bug.cgi?id=201817 )

epsilon-0, 2019-01-17:
> 
> 
> I just wanted to add, that for me everything still works (UEFI **not** legacy boot):
> 
> BIOS: F.31
> Notebook: HP ENVY x360 13-ag0004ng
> Kernels: 4.20.0-arch1, 4.20.3-arch1 (which is almost the mainline kernel)
> 
> It just doesn't work if I booted windows right before linux (as some suggest in the kernel bugtracker that this might be a problem:
> https://bugzilla.kernel.org/show_bug.cgi?id=201817 )

Installed 4.20.3 kernel and enabled disabled Legacy boot.
The touchscreen still not working...
I am using gentoo linux, with gentoo sources 4.20.3.
My config file:
<details><summary>File <summary>
<p>

#
# Automatically generated file; DO NOT EDIT.
# Linux/x86 4.20.3-gentoo Kernel Configuration
#

#
# Compiler: gcc (Gentoo 7.3.0-r3 p1.4) 7.3.0
#
CONFIG_CC_IS_GCC=y
CONFIG_GCC_VERSION=70300
CONFIG_CLANG_VERSION=0
CONFIG_IRQ_WORK=y
CONFIG_BUILDTIME_EXTABLE_SORT=y
CONFIG_THREAD_INFO_IN_TASK=y

#
# General setup
#
CONFIG_INIT_ENV_ARG_LIMIT=32
# CONFIG_COMPILE_TEST is not set
CONFIG_LOCALVERSION=""
# CONFIG_LOCALVERSION_AUTO is not set
CONFIG_BUILD_SALT=""
CONFIG_HAVE_KERNEL_GZIP=y
CONFIG_HAVE_KERNEL_BZIP2=y
CONFIG_HAVE_KERNEL_LZMA=y
CONFIG_HAVE_KERNEL_XZ=y
CONFIG_HAVE_KERNEL_LZO=y
CONFIG_HAVE_KERNEL_LZ4=y
# CONFIG_KERNEL_GZIP is not set
# CONFIG_KERNEL_BZIP2 is not set
# CONFIG_KERNEL_LZMA is not set
# CONFIG_KERNEL_XZ is not set
# CONFIG_KERNEL_LZO is not set
CONFIG_KERNEL_LZ4=y
CONFIG_DEFAULT_HOSTNAME="epicNOOB"
CONFIG_SWAP=y
CONFIG_SYSVIPC=y
CONFIG_SYSVIPC_SYSCTL=y
# CONFIG_POSIX_MQUEUE is not set
# CONFIG_CROSS_MEMORY_ATTACH is not set
# CONFIG_USELIB is not set
CONFIG_AUDIT=y
CONFIG_HAVE_ARCH_AUDITSYSCALL=y
CONFIG_AUDITSYSCALL=y
CONFIG_AUDIT_WATCH=y
CONFIG_AUDIT_TREE=y

#
# IRQ subsystem
#
CONFIG_GENERIC_IRQ_PROBE=y
[... 3556 lines cut ...]
# CONFIG_CPA_DEBUG is not set
# CONFIG_OPTIMIZE_INLINING is not set
# CONFIG_DEBUG_ENTRY is not set
# CONFIG_DEBUG_NMI_SELFTEST is not set
# CONFIG_X86_DEBUG_FPU is not set
# CONFIG_PUNIT_ATOM_DEBUG is not set
CONFIG_UNWINDER_ORC=y
# CONFIG_UNWINDER_FRAME_POINTER is not set
# CONFIG_UNWINDER_GUESS is not set

#
# Gentoo Linux
#
CONFIG_GENTOO_LINUX=y
CONFIG_GENTOO_LINUX_UDEV=y
CONFIG_GENTOO_LINUX_PORTAGE=y

#
# Support for init systems, system and service managers
#
CONFIG_GENTOO_LINUX_INIT_SCRIPT=y
# CONFIG_GENTOO_LINUX_INIT_SYSTEMD is not set

</p>
</details>

Eusebius1920, 2019-01-17:
After I tested the windows-boot thing again, the touchscreen no longer works d'oh .
(Even with 4.19.3 kernel downgrade)

Now I am getting this IRQ 7 error again...

epsilon-0, 2019-01-17:
> 
> 
> After I tested the windows-boot thing again, the touchscreen no longer works d'oh .
> (Even with 4.19.3 kernel downgrade)
> 
> Now I am getting this IRQ 7 error again...
> 
> ***Edit:**
> I could solve it for me:
> Enable legacy --> boot in linux --> disable legacy (and secure boot) --> boot in linux made touchscreen working for me again. If I boot in windows inbetween the touchscreen stops working.

WTH is this...
HP please fix >.>

jzkmath, 2019-06-29:
I have an HP envy x360 15m-cp0012dx and I found that touch does not work if you install ubuntu while in legacy mode. Reinstalled (19.04) in UEFI mode and both touch and the pen work, with a few minor issues:
The pen is seen as a mouse instead of a pen (which would turn the pointer into a plus sign if it did, but it doesn't). When I use the pen, it gives a "Keyboard Battery low" notification. 
Also, screen rotation doesn't work, but that might be unrelated.

Eusebius1920, 2019-07-01:
@jzkmath Screen Rotation should be related to this: https://bugzilla.redhat.com/show_bug.cgi?id=1651886

Quote:

> According to Alex from AMD, the driver for the Sensor Fusion HUB will be available around August

mmikel, 2019-07-05:
with this kernel touchscreen and stilys works again :)

https://kernel.ubuntu.com/~kernel-ppa/mainline/v5.2-rc7/

(KDE neon, HP ENVY x360 - 13-ag0004ur, Version: F.35, Release Date: 04/02/2019, BIOS Revision: 15.35, Firmware Revision: 92.46, Secure boot: disabled)

luyatshimbalanga, 2019-07-08:
> The pen is seen as a mouse instead of a pen (which would turn the pointer into a plus sign if it did, but it doesn't). 

From a terminal, type `libwacom-list-local-devices` to view a summary of tablet file. Edit _/usr/share/libwacom/<device>.tablet_ by additing the following line 
`Styli=0xfffff;0xffffe;0xffffd` below `IntegratedIn=Display;System;` .

Should it work, suggest the change to the developers.

> When I use the pen, it gives a "Keyboard Battery low" notification.

It is currently [a common issue](https://github.com/linuxwacom/input-wacom/wiki/Common-Issues).

jzkmath, 2019-07-08:
> > The pen is seen as a mouse instead of a pen (which would turn the pointer into a plus sign if it did, but it doesn't).
> 
> From a terminal, type `libwacom-list-local-devices` to view a summary of tablet file. Edit _/usr/share/libwacom/.tablet_ by additing the following line
> `Styli=0xfffff;0xffffe;0xffffd` below `IntegratedIn=Display;System;` .
> 
> Should it work, suggest the change to the developers.
> 
> > When I use the pen, it gives a "Keyboard Battery low" notification.
> 
> It is currently [a common issue](https://github.com/linuxwacom/input-wacom/wiki/Common-Issues).

I added this to my setup. Problem still exists. Still appears as a mouse instead of a stylus and gives the low battery notification. Is there a way to find out what values for "Styli" I need to put in the .tablet file?

Attached is my .tablet file.
[elan-2650.zip](https://github.com/linuxwacom/wacom-hid-descriptors/files/3369208/elan-2650.zip)

luyatshimbalanga, 2019-07-09:
> I added this to my setup. Problem still exists. Still appears as a mouse instead of a stylus and gives the low battery notification. Is there a way to find out what values for "Styli" I need to put in the .tablet file?

Currently there is not workaround about the low battery notification as highlighted on the linked common issue. As for the mouse pointer, does it occur within an application? The plus cursor displays on Gnome Wayland either in Overview or the top bar.
See [libwacom.stylus](https://github.com/linuxwacom/libwacom/blob/master/data/libwacom.stylus) to try the possible values for "Styli". It looks like it is possible to use some wacom pen value without an issue.

> 
> Attached is my .tablet file.
> [elan-2650.zip](https://github.com/linuxwacom/wacom-hid-descriptors/files/3369208/elan-2650.zip)

By the name, it looks like a new model which needs to be added in[ libwacom data]([(https://github.com/linuxwacom/libwacom/tree/master/data).

[12 more replies]