linuxwacom/wacom-hid-descriptors #14: X1 Yoga Gen 2. Pen and Mouse

sn2294, 2017-12-19:
[udevadm_0003:056A:50B6.0009.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/1571092/udevadm_0003.056A.50B6.0009.txt)
I installed the drivers to my X1 Yoga Gen 2 running Mint 18.3.  I use as default  left-hand mouse. The touch screen scrolls, but I could not mark or draw with the Pen in Xournal or Foxit Reader. 
I switched to right hand mouse, it works.
But after switching back to left  hand, and back to right, it would not work without a reboot.
[udevadm_0003:056A:50B6.0008.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/1571091/udevadm_0003.056A.50B6.0008.txt)

jigpu, 2017-12-19:
Please upload the full sysinfo tarball here -- its hard to get an overall picture of the hardware and software from only one or two logs.

sn2294, 2017-12-28:
[sysinfo.tkM70wemeh.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/1592181/sysinfo.tkM70wemeh.tar.gz)

jigpu, 2018-01-02:
Thanks! I've pushed this data to the repository as commit 4c061787584afe32aefc1054225df62d2272e435. I'll be creating a patch for libwacom shortly as well.

closed 2018-01-02 by jigpu