linuxwacom/wacom-hid-descriptors #18: ThinkPad S1 Yoga

ragtag, 2018-01-07:
ThinkPad S1 Yoga (model nr. 20DLCTO1WW) running Ubuntu Gnome 17.10, with Wayland, so the output from some of the xinput scripts is incomplete. If these are needed, I can boot from a USB to older Linux version, and re-run the script.

The following things don't work on Ubuntu 17.10.
- Doesn't disable touchpad or the red nipple in tent/tablet mode.
- No palm rejection when using the pen.

"xinput --list" does output the following:
```
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ xwayland-pointer:13                     	id=6	[slave  pointer  (2)]
⎜   ↳ xwayland-relative-pointer:13            	id=7	[slave  pointer  (2)]
⎜   ↳ xwayland-stylus:13                      	id=10	[slave  pointer  (2)]
⎜   ↳ xwayland-eraser:13                      	id=11	[slave  pointer  (2)]
⎜   ↳ xwayland-cursor:13                      	id=12	[slave  pointer  (2)]
⎜   ↳ xwayland-touch:13                       	id=9	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ xwayland-keyboard:13                    	id=8	[slave  keyboard (3)]
```
But the Absolute Proximity state doesn't change when running 'xinput query-state "xwayland-stylus:13" | grep Proximity' and putting a stylus to the screen.

I was going to update spin.py https://github.com/ragtag/spin to work with Ubuntu 17.10, but it  seems like it's going to be a bit problematic. A lot of the stuff spin.py does, is now done automatically (screen, tablet and touch screen rotation, and rotate lock), while some of the stuff, like palm rejection, no longer works. Stylus pressure does works in Gimp, Krita and MyPaint.

Anyway, I've attached my system info. If there is a simple way to detect the stylus proximity in Wayland, please give me a heads up, so I can write a simple script for palm rejection for my ThinkPad.

[sysinfo.zhFD5PcUe6.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/1610051/sysinfo.zhFD5PcUe6.tar.gz)

jigpu, 2018-01-08:
Thanks for the sysinfo output -- I've pushed it to the repository as commit 237f760122012e615233416a5cde1414ece6de0b.

You'll want to file two bugs at https://bugs.freedesktop.org/ against the "Wayland" product so that these issues can be fixed. The first bug should be against Wayland's "libinput" component, which should be disabling the touchpad and trackpoint when the device is in tent / tablet mode. The second bug should be against Wayland's "XWayland" component, which should be changing the proximity status. For future reference, would you mind posting a link to both of the bugs here after creating them?

closed 2018-01-08 by jigpu

ragtag, 2018-01-17:
Thank you for the reply.

Not disabling the trackpad bug report can be found here https://bugs.freedesktop.org/show_bug.cgi?id=104679

And the palm rejection one here https://bugs.freedesktop.org/show_bug.cgi?id=104680