linuxwacom/wacom-hid-descriptors #21: HP Spectre x360 (Kaby Lake 2017)

jcfergus, 2018-03-15:
Stylus works in landscape mode, but is not detected by gnome-control-center, and does not rotate with the screen and touchscreen when rotated.  

Sysinfo attached.  I upgraded to libwacom-26.1 from Ubuntu Bionic, but no change.  

[sysinfo.tNDvQdhhxL.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/1817237/sysinfo.tNDvQdhhxL.tar.gz)

jcfergus, 2018-03-16:
Tried updating to HEAD (0.29) - behavior remains the same.  It is detected by libwacom-list-local-devices:  

```
jferg@quartz ~ $ libwacom-list-local-devices 
[Device]
Name=Wacom Serial Tablet WACf004
DeviceMatch=serial:0000:0000;
Class=ISDV4
Width=0
Height=0
IntegratedIn=Display;System;
Styli=0xfffff;0xffffe;

[Features]
Reversible=false
Stylus=true
Ring=false
Ring2=false
Touch=false
TouchSwitch=false
StatusLEDs=
NumStrips=0
Buttons=0
---------------------------------------------------------------
```

but not by `xsetwacom --list devices` or by gnome-control-center.  In addition, when the screen is rotated, the stylus coordinates remain wrong.

jcfergus, 2018-03-16:
Finally found a message:
```
Mar 15 21:49:47 quartz /usr/lib/gdm3/gdm-x-session[31909]: (II) event21 - (II) ELAN0732:00 04F3:22E2 Pen: (II) device removed
Mar 15 21:50:42 quartz /usr/lib/gdm3/gdm-x-session[31909]: (II) event21 - (II) ELAN0732:00 04F3:22E2 Pen: (II) is tagged by udev as: Tablet
Mar 15 21:50:42 quartz /usr/lib/gdm3/gdm-x-session[31909]: (II) event21 - (II) ELAN0732:00 04F3:22E2 Pen: (II) tablet 'ELAN0732:00 04F3:22E2 Pen' unknown to libwacom
Mar 15 21:50:42 quartz /usr/lib/gdm3/gdm-x-session[31909]: (II) event21 - (II) ELAN0732:00 04F3:22E2 Pen: (II) device is a tablet
```

jcfergus, 2018-03-16:
Ok, got it working by adding in this tablet config:

hpspectrex360.tablet:
----------------------------
```
[Device]
Name=ELAN0732:00 04F3:22E2 Pen
DeviceMatch=i2c:04f3:22e2
IntegratedIn=Display;System
Styli=0xfffff;0xffffe

[Features]
Stylus=true
Reversible=false
Touch=false
TouchSwitch=false
Ring=false
Ring2=false
NumStrips=0
Buttons=0
```

jigpu, 2018-03-16:
Good to see that you figured it out :) I've just filed a bug for the "libwacom" project to add a tablet file similar to what you've provided.

The sysinfo.sh script didn't detect your Elan device, so I just pushed some modifications which should allow it to recognize it. Would you mind downloading/running it again so we can get data on it?

ebrensi, 2019-05-02:
@jcfergus, can you tell me where this `hpspectrex360.tablet` file goes?  I looked around online and can't find anything about it.

arneko, 2019-05-02:
Hello  @ebrensi,

I am also dabbling with this on Archlinux, and I believe this config is supposed to go to /usr/share/libwacom

However, there already seems to be a config file for my touchscreen. Can't get it to properly work though. Pen seems to work, including pressure, but rotating the screen does not reconfigure it. Also, I am under the impression, that it is supposed to show as a graphic tablet - it does not. Will try a bit more, before I will consider filing a bug report though. (not sure who'd be responsible yet)

ebrensi, 2019-05-02:
If it helps, I made a script to auto-orient the pen (stylus) device on the HP Spectre x360 Convertible.
https://gist.github.com/ebrensi/6d270bd84f04fbe0366fbbc3e5cf7054

It works when running directly from a terminal but for some reason it doesn't work as a systemd service.

jzkmath, 2019-06-13:
I don't know if this would be useful, but here is a config that I generated for my HP Spectre x360 15-BL012DX, which is the 2017 Kaby Lake version.
I have been trying to get my pen input to work on this machine, and there seems to be little to no information on this. Everyone pretty much says it works out of the box, so something is wrong here. 
Using Ubuntu 19.04...

[sysinfo.jRndX4Agef.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/3287563/sysinfo.jRndX4Agef.tar.gz)

jigpu, 2019-06-18:
@jzkmath It looks like your device is slightly different (and, importantly, has a different sensor) than the X360 this was originally filed for. Try using the following modified version of the above "hpspectrex360.tablet" file:

~~~
[Device]
Name=ELAN0732:00 04F3:22CA Pen
DeviceMatch=i2c:04f3:22ca
IntegratedIn=Display;System
Styli=0xfffff;0xffffe

[Features]
Stylus=true
Reversible=false
Touch=false
TouchSwitch=false
Ring=false
Ring2=false
NumStrips=0
Buttons=0
~~~

jzkmath, 2019-06-18:
> @jzkmath It looks like your device is slightly different (and, importantly, has a different sensor) than the X360 this was originally filed for. Try using the following modified version of the above [...]

Well I found out the issue. It turns out my pen's battery was dead, which rarely happens. Replacing it caused the pen to function normally!
Kinda feel really dumb about this, my apologies!

jigpu, 2019-11-19:
Closing pending updated sysinfo output from an x360 with the 04f3:22e2 sensor.

closed 2019-11-19 by jigpu