linuxwacom/wacom-hid-descriptors #22: hid descriptors for Cintiq Pro 16

IIerpos, 2018-03-26:
[wacom_Cintiq_Pro16_hid.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/1848728/wacom_Cintiq_Pro16_hid.tar.gz)

Also, I was wondering if anyone could help me figure out why I can't use full 4K resolution in Linux. It works fine in Windows, so it isn't a hardware limitation issue. If you don't have time or resources for that, no worries.

jigpu, 2018-03-27:
Thanks :) I've added this device to the descriptor list.

As to your question about 4K support, I'm not really sure. I don't have a system capable of driving 4K resolutions, so I've never tested it. Is the mode listed when you run `xrandr`? (I'll close this issue, but feel free to reply).

closed 2018-03-27 by jigpu

IIerpos, 2018-03-28:
I don't expect the problem is as easy to solve as some, and I don't even know whether you all are the ones to solve it, as the display and the pen capture are pretty separate animals, hardware wise. Nonetheless, I'll give you all the info I have, so you can do what you may when you may.

There have been a couple of times when I have had it running in full 4K resolution. But if the display is interrupted for any reason - logging in, touching the "tablet mode" button, powering off and on, etc - it gets booted down to 2560x1440. Then, sometimes, while running at 2560x1440, the display settings utility reports the ability to run it at 3840x2160, and then fails when you try to change resolutions (usually crashing the settings utility). It behaves the same whether I use Thunderbolt 3 or DisplayPort via the Wacom Link adapter.

In my brief foray into `xrandr` and attempts to set up custom resolutions and such, I found that `xrandr` would occasionally list the resolution, (3840x2160, 60fps) and then promptly _stop_ listing it if you tried to switch to it. There were no problems with fidelity when you actually were in 4K, however.

jigpu, 2018-03-29:
Thanks for the additional information. Since I'm only vaguely familiar with the graphics stack, your best bet would be to get in touch with the developers for your system's graphics drivers. They would have a better understanding of what could cause the 4K option to disappear from xrandr or only intermittently work. The following links should give you an idea of how to contact the various projects:

- [xf86-video-amdgpu](https://lists.freedesktop.org/mailman/listinfo/amd-gfx)
- [xf86-video-intel](https://01.org/linuxgraphics/community/xf86-video-intel)
- [xf86-video-nouveau](https://nouveau.freedesktop.org/wiki/)
- [nVidia Binary Driver](https://forums.geforce.com/default/board/33/geforce-drivers/)

You could also potentially try booting your system off of a couple of different live CDs (different distros, and different versions of the distros). If some distributions (or an older version of a specific distribution) are able to reliably run the monitor at 4K, that might give a hint that something in the graphics stack has recently broken.