linuxwacom/wacom-hid-descriptors #23: Samsung ATIV Book8 870Z5E

alanmm, 2018-03-31:
Wacom tablet model:
Bamboo CTH-470
(Doesn't move on Elementary OS)

[sysinfo.ZSsZRvwMAD.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/1865939/sysinfo.ZSsZRvwMAD.tar.gz)

jigpu, 2018-04-02:
Odd, I don't see anything in the sysinfo that would make me suspect that something is broken. The "xsetwacom" command detects the full complement of stylus, eraser, touch, and pad devices.

Please try using the [capture.sh script](https://gist.github.com/jigpu/b7b47e4bede584cec8562f666fd84af4). Select the "stylus" device when prompted, tap the tablet a few times with the stylus, hit CTRL+C, and upload the generated tarball.

jigpu, 2018-07-06:
Closing due to lack of necessary information. Please feel free to reply with the requested information.

closed 2018-07-06 by jigpu