linuxwacom/wacom-hid-descriptors #26: HP Pavilion x360 convertible 15 br0xx. Pen/stylus works in Ubuntu 16.04, but broken in 18.04

fndercole, 2018-05-01:
I have a HP Pavilion x360 convertible 15 br0xx. This laptop has a multitouch touchscreen and an active pen/stylus (non-wacom).

On ubuntu 16.04, the touchscreen used with fingers and the pen/stylus work properly. The touchscreen was identified as «ELAN0732:00 04F3:24B8» as slave pointer and the pen as «ELAN0732:00 04F3:24B8 Pen» as slave pointer too.

But, on ubuntu 18.04 bionic with gnome shell, the pen/stylus is identified «ELAN0732:00 04F3:24B8 Pen» as slave keyboard instead as slave pointer. Touchscreen work well until pen/stylus is used. At this point, a new device «ELAN0732:00 04F3:24B8 Pen Pen (0)» is added to pointers list. The touchscreen stops working and the pen/styles works but with some issues.
```
    ~$ xinput --list
    ⎡ Virtual core pointer id=2                  [master pointer (3)]
    ⎜ ↳ Virtual core XTEST pointer id=4          [slave pointer (2)]
    ⎜ ↳ ETPS/2 Elantech Touchpad id=15           [slave pointer (2)]
    ⎜ ↳ ELAN0732:00 04F3:24B8 id=10              [slave pointer (2)]
    ⎜ ↳ ELAN0732:00 04F3:24B8 Pen Pen (0) id=18  [slave pointer (2)]
    ⎣ Virtual core keyboard id=3                 [master keyboard (2)]
        ↳ Virtual core XTEST keyboard id=5       [slave keyboard (3)]
        ↳ Power Button id=6                      [slave keyboard (3)]
        ↳ Video Bus id=7                         [slave keyboard (3)]
        ↳ Power Button id=8                      [slave keyboard (3)]
        ↳ HP Wide Vision HD Camera: HP Wi id=9   [slave keyboard (3)]
        ↳ Intel Virtual Button driver id=12      [slave keyboard (3)]
        ↳ Intel Virtual Button driver id=13      [slave keyboard (3)]
        ↳ AT Translated Set 2 keyboard id=14     [slave keyboard (3)]
        ↳ HP WMI hotkeys id=16                   [slave keyboard (3)]
        ↳ HP Wireless hotkeys id=17              [slave keyboard (3)]
        ↳ ELAN0732:00 04F3:24B8 Pen id=11        [slave keyboard (3)]
```

After disable devices id=11 and id=18, touchscreen recovers its multitouch feature.
There is no combination enabling/disabling ids 10, 11 and 18, to get multitouch and pen working together.

On Wayland session, I get the same behavior.

Note: Ubuntu 18.04 LTS clean installation with «minimal install option» and no other customisation.

[sysinfo.wdcIudZogZ.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/1965141/sysinfo.wdcIudZogZ.tar.gz)

jigpu, 2018-05-08:
Thanks for the sysinfo output. I've added it to the repository so we have the information on this device in the future. Our driver isn't used for the ELAN sensors, so I unfortunately can't be too much help with fixing your problem. I would suggest the following:

1. Register for an account at https://bugs.freedesktop.org/ and select "file a bug"
2. Select the "xorg" product
3. Choose the "Input/libinput" component at top
4. Fill in the bug details, including a link back to this bug so people can read the sysinfo output

Just based off of your description, it sounds like either the kernel driver has started exposing more devices and is confusing the userspace "libinput" driver, or some recent change to libinput has broken support for your device. The libinput developers at freedesktop would know best and should be able to properly diagnose what's going on. Sorry we can't be more help!

closed 2018-05-08 by jigpu

fndercole, 2018-05-10:
Thanks you, for your advice!
Best.