linuxwacom/wacom-hid-descriptors #28: Acer Spin 11 CP311 Wacom EMR Pen Not Working

N1nDr0id, 2018-05-29:
As described in this [Reddit post I made](https://www.reddit.com/r/Crouton/comments/8mvwux/got_an_acer_spin_11_w_wacom_emr_pen_recently_and/) , /u/jigpu mentioned I should post this on GitHub along with a tar.gz file generated from a sysinfo.sh file he gave me.
[sysinfo.PcR6J83Fkz.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/2051561/sysinfo.PcR6J83Fkz.tar.gz)

Hopefully this will be able to help you guys out.

N1nDr0id, 2018-05-29:
@jigpu Hopefully this will help a bit.

jigpu, 2018-05-30:
Thanks for the sysinfo @N1nDr0id :) It looks like our script had trouble grabbing some of the sensor's data, so I've made a few modifications to it. Would you mind getting a fresh copy of the script, re-running it, and uploading the new results?
Following up on the information from Reddit, it does indeed appear that your device is using the generic driver rather than the Wacom-specific one. Right now there's no good way to change that, so we'll have to look at changing the name through the X driver instead or have Krita change its behavior.

N1nDr0id, 2018-05-30:
Here you are, updated version. Made sure to run it under sudo as well :) 
[sysinfo.6DRRc1MlrI.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/2055767/sysinfo.6DRRc1MlrI.tar.gz)

N1nDr0id, 2018-05-30:
And also, JUST in case, here's that same updated info, but the file was NOT run under sudo.
[sysinfo.g9gsMERJT1.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/2055786/sysinfo.g9gsMERJT1.tar.gz)