linuxwacom/wacom-hid-descriptors #30: Support for Wacom Cintiq 15x PL-550

Lvaskz, 2018-06-24:
Hello, I attached file to give support to the mentioned tablet. Ubuntu 18.04 does not recognize the tablet and I can not calibrate or draw anything. In linuxmint 18.03 works perfectly.
[sysinfo.3XqU88nvax.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/2131201/sysinfo.3XqU88nvax.tar.gz)

Lvaskz, 2018-06-24:
Hi, with linuxmint 19 beta,  wacom Cintiq 15x  Pl 550 is no recognized

jigpu, 2018-06-26:
Hi there -- could you let me know if your Cintiq is connected over USB or serial? The *only* mention of it in your logs is as a monitor, not as an input device. It isn't listed as a USB device by `lsusb`, the kernel's `dmesg` log doesn't show any interesting devices, and the `xsetwacom` and `xinput` tools don't list it either.

If you're connected over serial, I would strongly suggest using USB instead. The serial connection is much harder to get set up properly, which could be why it suddenly stopped working (a minor config file difference between Ubuntu and Mint could easily be the problem).

If you're connected over USB, more in-depth troubleshooting is required. Does the tablet's green status LED light when touching the pen to the screen?

Lvaskz, 2018-06-26:
Wacom cintiq 15x is connected via usb.

"Does the tablet's green status LED light when touching the pen to the screen?" Yes, it does

jigpu, 2018-06-26:
Hmm, okay. In that case, something very strange is going on. The tablet is clearly working enough to light the status LED, but the syslog didn't indicate the kernel ever saw a USB device. Please try running `dmesg -w` to "watch" the kernel logs in real-time. Take note of the timestamp of the last message, and then unplug and reconnect the Cintiq's USB cable. Were any new messages logged? Please copy any new messages here.

Lvaskz, 2018-06-26:
I have something to say. My cintiq has a cable to conect to my computer via hdmi adapter and a usb cable to the computer for the pen. I add a manual pdf, watch page 11 to see the tablet if that helps 
[Cintiq15X-User-Manual.pdf](https://github.com/linuxwacom/wacom-hid-descriptors/files/2137986/Cintiq15X-User-Manual.pdf)

Lvaskz, 2018-06-29:
[    0.000000] microcode: microcode updated early to revision 0x23, date = 2018-01-18
[    0.000000] Linux version 4.15.0-23-generic (buildd@lgw01-amd64-055) (gcc version 7.3.0 (Ubuntu 7.3.0-16ubuntu3)) #25-Ubuntu SMP Wed May 23 18:02:16 UTC 2018 (Ubuntu 4.15.0-23.25-generic 4.15.18)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.15.0-23-generic root=UUID=9ec20f28-0305-4600-8ed3-22c7465ef553 ro quiet splash vt.handoff=1
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x000000000009efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000cadc4fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000cadc5000-0x00000000cadcbfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000cadcc000-0x00000000cb1f4fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000cb1f5000-0x00000000cb7f0fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000cb7f1000-0x00000000da891fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000da892000-0x00000000dabdafff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000dabdb000-0x00000000dabf5fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000dabf6000-0x00000000db571fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000db572000-0x00000000dbf6ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000dbf70000-0x00000000dbffdfff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000dbffe000-0x00000000dbffefff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dd000000-0x00000000df1fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000011fdfffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.31 by American Megatrends
[    0.000000] efi:  ACPI=0xdabe0000  ACPI 2.0=0xdabe0000  SMBIOS=0xf04c0  MPS=0xfd820 
[    0.000000] secureboot: Secure boot could not be determined (mode 0)
[    0.000000] random: fast init done
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: Dell Inc. Inspiron 3442/0RW1RW, BIOS A05 11/14/2014
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x11fe00 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-CFFFF write-protect
[    0.000000]   D0000-DFFFF uncachable
[    0.000000]   E0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 0000000000 mask 7F00000000 write-back
[    0.000000]   1 base 0100000000 mask 7FE0000000 write-back
[    0.000000]   2 base 00E0000000 mask 7FE0000000 uncachable
[    0.000000]   3 base 00DE000000 mask 7FFE000000 uncachable
[    0.000000]   4 base 00DD000000 mask 7FFF000000 uncachable
[    0.000000]   5 base 011FE00000 mask 7FFFE00000 uncachable
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000]   8 disabled
[    0.000000]   9 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000000] total RAM covered: 4046M
[    0.000000] Found optimal setting for mtrr clean up
[    0.000000]  gran_size: 64K 	chunk_size: 64M 	num_reg: 7  	lose cover RAM: 0G
[    0.000000] e820: update [mem 0xdd000000-0xffffffff] usable ==> reserved
[    0.000000] e820: last_pfn = 0xdbfff max_arch_pfn = 0x400000000
[... 841 lines cut ...]
[ 1757.627615] wlp6s0: RX AssocResp from c0:56:27:27:93:cd (capab=0x411 status=0 aid=2)
[ 1757.627725] wlp6s0: associated
[ 1757.650879] IPv6: ADDRCONF(NETDEV_CHANGE): wlp6s0: link becomes ready
[ 1757.807339] atkbd serio0: Unknown key pressed (translated set 2, code 0x8d on isa0060/serio0).
[ 1757.807342] atkbd serio0: Use 'setkeycodes e00d <keycode>' to make it known.
[ 1760.889226] usb 1-1.6: device descriptor read/64, error -110
[ 1771.414262] usb 1-1.6: New USB device found, idVendor=0cf3, idProduct=0036
[ 1771.414268] usb 1-1.6: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 7646.680414] xhci_hcd 0000:00:14.0: Root hub is not suspended
[ 7646.915841] usb 2-1: new low-speed USB device number 2 using xhci_hcd
[ 7647.043961] usb 2-1: device descriptor read/64, error -71
[... cut ...]
[ 7648.640295] usb 2-1: Device not responding to setup address.
[ 7648.848285] usb 2-1: Device not responding to setup address.
[ 7649.055775] usb 2-1: device not accepting address 4, error -71
[ 7649.183674] usb 2-1: new low-speed USB device number 5 using xhci_hcd
[ 7649.184078] usb 2-1: Device not responding to setup address.
[ 7649.392249] usb 2-1: Device not responding to setup address.
[ 7649.599772] usb 2-1: device not accepting address 5, error -71
[ 7649.599858] usb usb2-port1: unable to enumerate USB device

Lvaskz, 2018-06-29:
I'm going to close this. The tablet is damaged in some of its cables and is no longer recognized in mint 18.3. However, I would like to know if the new Wacom Intous S, is supported?

closed 2018-06-29 by Lvaskz

jigpu, 2018-07-02:
Hmm, yeah, error -71 doesn't look good for a USB device...

All three of the recent generations of "Intuos S" tablets should be supported.

Lvaskz, 2018-07-02:
Ok, thanks for all your help.