linuxwacom/wacom-hid-descriptors #32: HP Spectre x360 Convertible 13-ac0XX

cuchaz, 2018-07-21:
Hello,

I'm trying to resolve issues with the touchscreen and stylus on my HP convertible laptop.

1. The coordinate system for the stylus doesn't appear to rotate with the screen.
2. Drawing with the stylus (in eg xournal, gimp) often causes extra lines to appear where they shouldn't. It's a bit hard to explain, but starting a touch with the stylus causes an extra line to be drawn to the touch position from the current cursor position (the cursor lags behind the physical stylus location quite a bit when moving quickly).  Maybe it's an issue with events from multiple pointers getting blended together in the wrong order somehow? I'm not really sure what's going on there. I've tried lots of drawing apps and they all have the same behavior though, so I'm guessing the issue is lower-level than the apps themselves.

The touchscreen is an ELAN 0732 04F3:24DB. It looks similar to touchscreens that were added to libwacom recently. I tried making a .tablet file for it, but other than `libwacom-list-local-devices` showing it (along with a "serial" wacom tablet), adding that tablet file doesn't seem to resolve either of the two issues I'm seeing.

The sysinfo results should be attached. I ran the sysinfo script after adding the .tablet file.
[sysinfo.M4NrMxnhrX.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/2216506/sysinfo.M4NrMxnhrX.tar.gz)

If you need any more info, let me know. I'm happy to help. And thanks in advance. =)

cuchaz, 2018-07-21:
Update: I managed to workaround issue 2 by switching from the libinput driver to the evdev driver.  Hooray! There must be some kind of issue in libinput that causes the unwanted extra lines when drawing with the stylus. Issue 1 still persists with the evdev driver though.

jigpu, 2018-07-23:
I've pushed the sysinfo data up to the repository, thanks!

Glad to hear that you were able to get issue 2 sorted out, though I would highly recommend filing a bug at https://gitlab.freedesktop.org/libinput/libinput/issues/ to let the libinput developers know about the issue. The evdev driver doesn't exist in the world of Wayland, so those extra lines may re-appear in the future if/when you switch to a Wayland-based desktop.

Automatic rotation is a feature of the desktop environment. I think it only works with GNOME and desktops based on it. If you're using such an environment, you might see if auto-rotation starts to work when using xf86-input-wacom instead (see [this post](https://www.reddit.com/r/linuxhardware/comments/9098wd/bamboo_ink_button_3_not_working/e2qubkz) for information on getting your Elan sensor to work with the wacom driver). If you're running something like KDE then you'll need to find/write a script to do the rotation for you.

closed 2018-07-23 by jigpu

cuchaz, 2018-07-23:
Awesome, thanks for the info! I didn't know where the libinput repo was. I'll let them know about the bug.

I considered making a script to handle the rotation issue. Linux Mint uses Gnome 3 I think, so I wasn't sure if its auto-rotation feature would conflict with a script. [This link](https://gist.github.com/mildmojo/48e9025070a2ba40795c) might be a good starting point for getting a script like that working though. I'll try the other driver first though.

Element-126, 2018-09-01:
@cuchaz I have the same issue (point 2.) as you on Fedora 28 (Gnome 3 Wayland). Did you report the bug ?

cuchaz, 2018-09-02:
@Element-126 [Yeah, and we fixed it too.](https://gitlab.freedesktop.org/libinput/libinput/issues/94)

Element-126, 2018-09-02:
@cuchaz Wow! Thanks a lot for the fix!

But something is weird. I am running a two-week-old 1.11.902 (which includes your fix) and still seeing the issue. So I may need to reopen the issue. I will first check if I also have the pixel discontinuity.

Also, did you see a precision drop when going from Xorg to Wayland ? It is quite pronounced on my machine, and I may need to file another issue for this.

cuchaz, 2018-09-02:
No problem! =)

I'm on Linux Mint 19, which is still using Xorg I think. I don't know anything about Wayland.

jigpu, 2018-09-04:
I'd put libinput as the most suspect component, @Element-126. In theory XWayland should only be wrapping the output of libinput, so I'd expect similar behavior from a GTK3 application. libwacom is only used to provide additional information to the GNOME control panel and won't have any effect on the pen behavior.

MyPaint versions 1.2 and later should use GTK3, and I've used it in the past for Wayland testing. I believe Fedora 28 ships a new enough version that you could test out.

Element-126, 2018-09-04:
@jigpu I can indeed reproduce the issue using MyPaint 1.2.1 (flathub version). So I will first check if the libwacom support changes anything (in case libinput takes a different code path for unsupported devices) and then I will tentatively report the issue to libinput.

Thanks for your help!