linuxwacom/wacom-hid-descriptors #33: Sysinfo - Xp Pen Star 06

jeffreyantony, 2018-07-23:
Hi,

I am attaching my Xp Pen Star 06 sysinfo file
[sysinfo.v8RUkqOWwA.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/2220625/sysinfo.v8RUkqOWwA.tar.gz)

I don't get any output when running xwacomset. How can I get this tablet supported ?

jigpu, 2018-07-23:
Your sysinfo output doesn't contain any useful information. The problem is that XP Pen is not a vendor that is recognized by the sysinfo script. You'll need to find the vendor ID of the tablet. Try posting the output of `lsusb`.

jeffreyantony, 2018-08-15:
Bus 002 Device 006: ID 28bd:0078  
Bus 002 Device 005: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 002 Device 004: ID 214b:7250  
Bus 002 Device 003: ID 090c:37b3 Silicon Motion, Inc. - Taiwan (formerly Feiya Technology Corp.) 
Bus 002 Device 002: ID 8087:0020 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0489:e00d Foxconn / Hon Hai Broadcom Bluetooth 2.1 Device
Bus 001 Device 002: ID 8087:0020 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

jigpu, 2018-08-21:
Hmm, not terribly obvious which device is which. Could you run `lsusb` again without the tablet connected for comparison? That should let us know what VID needs to be added to the sysinfo script.

jigpu, 2019-11-19:
Closing pending additional information.

closed 2019-11-19 by jigpu