linuxwacom/wacom-hid-descriptors #36: Lenovo Yoga 730-15ikb no pen input

organburner, 2018-08-21:
[sysinfo.sk3f2Cp2eT.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/2306346/sysinfo.sk3f2Cp2eT.tar.gz)

Hi,

I have a Lenovo Yoga 730-15ikb, it comes with a Lenovo Active pen 2 which I have been trying to get to work for a few days. I've had fedora and gnome installed and tried ubuntu with gnome and am currently using linux mint with cinnamon and I have been unable to get the pen working in any of these.

Since this is my first computer with a pen using AES tech I'm not sure how this device should work, bluetooth keeps showing the device as disconnected until I press the sync button, I don't know if this is the issue?
I don't know where the device should appear as lsusb and lspci gives me nothing pointing to a wacom device?
Xinput does list Wacom HID 516B devices though.
Also for whatever it's worth the touchscreen does work (But I'd prefer never using the touchscreen with my dirty filthy fingers if I could get the pen working)
Any help is appreciated.

jigpu, 2018-08-21:
Also, I noticed that your Xorg log file had the following line in it:
~~~
[   186.247] (EE) event8  - MSFT0001:02 06CB:7F8F Touchpad: kernel bug: Touch jump detected and discarded.
See https://wayland.freedesktop.org/libinput/doc/1.10.4/touchpad_jumping_cursor.html for details
~~~
This error doesn't have anything to do with the pen, but it sounds like a touchpad bug that you might want to report (instructions are in the link).

jigpu, 2018-08-21:
It turns out I may have been partially incorrect about the Bluetooth button... My original report of it not being supported was because of issues with the Bluetooth module that's used in Wacom-branded AES pens (e.g. their "Bamboo Ink"). I just realized that other AES pens might not have the same issue and could potentially work.

I was able to borrow an AES pen for an HP laptop and was able to see keyboard events being sent to the X server when I pressed the button. If your pen is Lenovo-branded rather than Wacom-branded, there's a chance that it might work as well. You might try pairing the pen, running `xinput test-xi2` to log all mouse/keyboard events sent by the X server, and then pressing the button. With any luck, you may see KeyPress and KeyRelease events logged in response.