linuxwacom/wacom-hid-descriptors #46: hp envy x360 15-cn0008ng

blauertee, 2019-01-15:
My stylus pen (ELAN0732) is detected and works, but I get weird jumping behavior (the pen draws straight lines from it's last position to it's next one in multiple drawing apps). 

[image]

I was redirected here from https://github.com/linuxwacom/input-wacom/issues/65

[sysinfo.ADroBI3Hdk.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/2758730/sysinfo.ADroBI3Hdk.tar.gz)

Thanks!

jigpu, 2019-01-15:
Thanks! I'll be pushing the data up to the repository shortly.

The logs indicate that your device is using the "hid-multitouch" kernel driver and the "xf86-input-libinput" Xorg driver. Also, I notice a lot of warnings like this in your kernel log: `i2c_hid i2c-ELAN0732:00: i2c_hid_get_input: incomplete report (67/65535)`. It sounds like the HID descriptor might have problems, which could possibly explain the odd behavior.

blauertee, 2019-01-18:
The jumps have been solved simply by upgrading to the newest libinput version :)

https://gitlab.freedesktop.org/libinput/libinput/issues/219

closed 2019-01-18 by blauertee