linuxwacom/wacom-hid-descriptors #48: Nuvision Solo 10 Draw

ShapeShifter499, 2019-02-12:
This tablet supports Microsoft Surface pens and I've personally been using mine with a Surface 4 pen. Attached is the sysinfo generated from the 'sysinfo.sh' script.
[sysinfo.RGjMc1uFhu.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/2857992/sysinfo.RGjMc1uFhu.tar.gz)

ShapeShifter499, 2019-02-12:
I should add, the pen appeared to work fine before I wiped Windows completely from this tablet. The eraser on the tip, the one button on the side, and the pressure sensitivity.

I'm now on the latest Antergos/Arch Linux and the pen works for standard mouse-like operations but doesn't work for drawing in Xournal. 

But if I add the device ID of '04f3:22f7' to /usr/share/X11/70-wacom.conf   it works in Xournal for drawing but not for erasing. Flipping my pen to the eraser side just triggers drawing even after changing Xournal settings.

jigpu, 2019-03-25:
Thanks -- added the tablet info to the database

closed 2019-03-25 by jigpu