linuxwacom/wacom-hid-descriptors #49: dell latitude 7390 2-in-1 (13)

lukyluket, 2019-03-05:
Here I attach the sysinfo for the Latitude, everything seems to work ok, though when in tablet mode, the pen randomly stops working I need revert to laptop mode and back to tablet mode in order to get it working again, sometime it is also reactivated by touching the screen with the fingers, but not alwaystm
[sysinfo.iwM2pJBDxQ.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/2932141/sysinfo.iwM2pJBDxQ.tar.gz)

jigpu, 2019-03-12:
Thanks, I've added the info to the database. If you have the ability to run Windows on the Latitude it would be interesting to know if it is also affected. Tablet mode can be a bit more challenging for the sensor because of the additional electrical noise, and some devices experience problems because of it.

If Windows works fine, you might consider opening a bug for the issue at https://github.com/linuxwacom/input-wacom/issues/

closed 2019-03-12 by jigpu

LaserTron, 2019-06-12:
> Here I attach the sysinfo for the Latitude, everything seems to work ok, though when in tablet mode, the pen randomly stops working I need revert to laptop mode and back to tablet mode in order to [...]

Hi, I'm running Manjaro and I'm completely unable to get the stylus to work. This is the only reference to 
 the Dell 7390 2in1 that I found. Could you direct me to instructions to get the pen workin?

p.s. I aplogize if I'm being inappropriate asking this here.

jigpu, 2019-06-12:
@LaserTron would you mind also running the [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and attaching the results to this issue? That should give me enough information about your system to help you further.

lukyluket, 2019-06-12:
In my case it works out of the box beside the few issues I have mentioned (
pen randomly stops writing and need to get battery out and in again to make
it work again)

LaserTron, 2019-06-13:
> @LaserTron would you mind also running the [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and attaching the results to this issue? That should give me enough information about your system to help you further.

Thanks alot. Here's the output.
[sysinfo.HALYVxjaCd.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/3283661/sysinfo.HALYVxjaCd.tar.gz)

LaserTron, 2019-06-13:
> In my case it works out of the box beside the few issues I have mentioned ( pen randomly stops writing and need to get battery out and in again to make it work again)
> […](#)

I see. If you don't mind, could you tell me which exact stylus model you are using? Thanks.

jigpu, 2019-06-18:
Thanks @LaserTron. It looks like your sensor is the same as @lukyluket has, and your system is able to detect and bring up the device. I would definitely suspect a stylus compatibility issue at the moment. Can you provide us with information about the stylus that you're using?

lukyluket, 2019-06-18:
Sorry for the late reply I guess mine should be
https://www.dell.com/en-us/shop/dell-active-pen-pn557w/apd/750-aaty/pc-accessories?ref=p13n_ena_pdp_vv&c=us&cs=19&l=en&s=dhs

lukyluket, 2019-06-18:
Bytheway I found this that solve my issues with the pen intermittently
writing
https://www.dell.com/support/article/us/en/19/sln314426/dell-active-pen-pn556w-pn557w-intermittent-writing?lang=en

LaserTron, 2019-06-19:
> Thanks @LaserTron. It looks like your sensor is the same as @lukyluket has, and your system is able to detect and bring up the device. I would definitely suspect a stylus compatibility issue at the moment. Can you provide us with information about the stylus that you're using?

Sure! The stylus I'm using is [the pn350M:](https://www.dell.com/pn350m).

LaserTron, 2019-06-19:
Actaully I just checked this: 
https://www.dell.com/support/article/bz/en/bzbsdt1/sln316630/dell-active-pen-pn350m-support-center?lang=en

Sorry for wasting your time :-( 
It's just that I expectedt the vendor to know what they were doing.

lukyluket, 2019-06-19:
Yes it seems that the inspiron 7390 is compatible but not the latitude,
sorry to hear that you have the wrong pen.

Regards
Luca

LaserTron, 2020-09-18:
Hi! Thanks for helping me last time. I eventually bought a compatible pen and it worked out of the box!!
Unfortunately the pen sporadically stops working, you mentioned the same issue initially. How did you work around it?

LaserTron, 2020-09-18:
> In my case it works out of the box beside the few issues I have mentioned ( pen randomly stops writing and need to get battery out and in again to make it work again)
> […](#)

@lukuluket,

Just like that. I took the battery out and put it back in the pen and now it appears to be working reliably. Is this a coincidence or has this been your experience as well?

lukyluket, 2020-09-18:
Hi there I also found that properly tightening the small battery
compartment (the one that powers the Bluetooth) helped to stop the lack of
functioning
Cheers
Luca

daveol, 2025-11-16:
I seem to have a different hardware revision with a `Wacom 4840` instead of a `Wacom 4841` in the earlier dumps, so here is my sysinfo dump too.

[sysinfo.6FI0oWHVlz.tar.gz](https://github.com/user-attachments/files/23567531/sysinfo.6FI0oWHVlz.tar.gz)