linuxwacom/wacom-hid-descriptors #53: Sysinfo - HP Spectre x360 13-40xx (ELAN usb:04f3:2072)

jkearneyma, 2019-05-07:
An older 2015 version of this series.  Stylus works as a mouse, but is not detected as a pen.

ELAN usb:04f3:2072

[sysinfo.5Gp29Fy1Mh.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/3154143/sysinfo.5Gp29Fy1Mh.tar.gz)

jkearneyma, 2019-05-08:
With the following .tablet file, gnome-control-center shows the tablet, but the stylus is still not detected or configurable.

# ELAN touchscreen/pen sensor present in the HP Spectre x360 Convertible 13-40xx
[Device]
Name=ELAN 2072
ModelName=
DeviceMatch=usb:04f3:2072
Class=PenDisplay
IntegratedIn=Display;System
Styli=0xffffd

[Features]
Stylus=true
Touch=true
Buttons=0

jigpu, 2019-05-20:
Thanks, I've gone ahead and added the data to the repository. I would suggest creating a pull request for [libwacom](https://github.com/linuxwacom/libwacom/) with your proposed tablet definition file so that others can benefit from it. I suspect the issue of the stylus not showing up in the control panel to be a well-known GNOME bug that has recently been fixed in version 3.32.

closed 2019-05-20 by jigpu