linuxwacom/wacom-hid-descriptors #59: Lenovo Miix 520-12IKB

orki, 2019-07-14:
Sysinfo attached.
[sysinfo.s4Ggr1usDS.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/3390297/sysinfo.s4Ggr1usDS.tar.gz)

orki, 2019-07-14:
With a .tablet file, I can get libwacom to display the device:
```
ravi@hewitt ~]$ libwacom-list-local-devices 
[Device]
Name=Wacom HID 513B Pen
DeviceMatch=i2c:056a:513b;
Class=Unknown
Width=10
Height=6
IntegratedIn=Display;System;
Styli=0xfffff;0xffffe;

[Features]
Reversible=false
Stylus=true
Ring=false
Ring2=false
Touch=false
TouchSwitch=false
StatusLEDs=
NumStrips=0
Buttons=0
```
but the device is still not seen by gnome as a pen. Also tried `Class=ISDV4` with same results.

jigpu, 2019-07-15:
Thanks for the data, I've added it to the repository and submitted a pull request for a tablet information file. Please try that tablet file in place of your own and let me know if it works any better. If the stylus issue is still present, please comment on the pull request so that changes can be made.

As a future note to myself, your distro appears to be new enough to have GNOME 3.32 and so shouldn't be affected by [this issue](https://bugzilla.gnome.org/show_bug.cgi?id=793499). You might, however, need the ID numbers of your specific stylus as outlined in [this issue](https://github.com/linuxwacom/libwacom/issues/57).

closed 2019-07-15 by jigpu