linuxwacom/wacom-hid-descriptors #61: Lenovo X1 Yoga (2019 - 4th Generation)

expoqeteer, 2019-08-08:
Please add full support for new X1 Yoga (4th gen).  Pen and touch work intermittently.  GNOME settings does not recognize the Wacom digitizer.

sysinfo file attached.  I redacted some sensitive information like usernames and hostnames.

Thanks!

[sysinfo.eFXsWQfXUf.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/3483074/sysinfo.eFXsWQfXUf.tar.gz)

expoqeteer, 2019-08-08:
Please let me know if there's any additional information I can provide or testing that I can perform.

jigpu, 2019-10-10:
Sorry for the delay; data has been added and a libwacom pull request created.

closed 2019-10-10 by jigpu

Hubbz, 2019-11-28:
Good day. Might need to reopen this as I'm unable to get pen interaction with the current drivers. Touch works flawlessly, but pen isn't recognized on the screen or in System > Devices > Wacom Tablet > Stylus.

I've successfully installed or updated the kernel driver, x driver and libwacom library to latest as indicated at https://linuxwacom.github.io/

If you need anything from me other than this file, just let me know.

[sysinfo.8cu8qTaKyb.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/3902427/sysinfo.8cu8qTaKyb.tar.gz)

jigpu, 2019-12-02:
Thanks for the sysinfo. It looks like your sensor has a different PID despite being the same model. I've added the info to the database and opened a pull request to add your sensor into libwacom as well. You can save this [`isdv4-51be.tablet`](https://raw.githubusercontent.com/linuxwacom/libwacom/85ddeb2966349207db53a024bfd23beb4a51b634/data/isdv4-51be.tablet) file into your `/usr/share/libwacom` directory to have your system recognize the sensor while you wait for an updated version of libwacom to be released.

The stylus won't be recognized in `System > Devices > Wacom Tablet > Stylus` at the moment (even with the tablet definition file in place). We are working on adding proper support for tablet PC styli to libwacom and GNOME.

You should be able to move the cursor around the screen with your pen even without any of these fixes. It sounds like that might not be the case for you, however? If you can't move the cursor or click on anything with the pen, please open an issue at https://github.com/linuxwacom/xf86-input-wacom/issues/ so we can figure out what is going wrong.

Hubbz, 2019-12-03:
Thank you @jigpu 

Yeah, I'm still having issues with my pen. I tested last night booting into Windows 10 instead and had the same result. Might be defective hardware... I've seen posts in other forums of people having the same issue. Anywho, I'll follow up where you've indicated.

Thanks again.

Hubbz, 2019-12-10:
Just a quick follow up for anyone who stumbles upon this. The pen I had was defective hardware which seems to be fairly common. I got a new one from Lenovo, charged it, and everything is working as it should.