linuxwacom/wacom-hid-descriptors #66: Lenovo X1 Yoga Gen 3 (056A:5147)

vozhyk-, 2019-10-14:
I've noticed that there's no mention of 056A:5147 here, only of 5146.

Pen and touch works. Gnome Settings doesn't recognize the stylus/tablet (https://gitlab.gnome.org/GNOME/gnome-control-center/issues/274#note_509140).

[sysinfo.sMNMfyxaJt.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/3723229/sysinfo.sMNMfyxaJt.tar.gz)

vozhyk-, 2019-10-14:
Copying `/usr/share/libwacom/isdv4-5146.tablet`:
```diff
--- /usr/share/libwacom/isdv4-5146.tablet       2018-10-25 02:42:49.414238852 +0200
+++ /usr/share/libwacom/isdv4-5147.tablet       2019-10-14 08:46:05.523326941 +0200
@@ -1,8 +1,8 @@
 # this is for the Wacom pen + touchscreen as found in the Lenovo ThinkPad X1 Yoga (3rd-gen)
 
 [Device]
-Name=Wacom ISDv4 5146
-DeviceMatch=usb:056a:5146
+Name=Wacom ISDv4 5147
+DeviceMatch=usb:056a:5147
 Class=ISDV4
 Width=12
 Height=7
```

 makes Gnome Settings show a tablet and 2 styli, both named `General Pen`.

I have only one stylus, with an eraser (lower) button and an upper button. The setting for the upper button's action works properly, but the lower button setting does nothing. The button settings for the "second stylus" also do nothing.

I've tried https://github.com/linuxwacom/libwacom/issues/12#issuecomment-531341160 (with a stylus ID of `0x11`, `HasEraser=true` and `Buttons=1`) and got similar results to https://github.com/linuxwacom/libwacom/issues/12#issuecomment-531569654 - Settings showed one setting for a button, but the setting did nothing. With `Buttons=2`, the behavior was the same as above. In both cases, the second stylus (`General Pen`) was still present.

jigpu, 2019-10-14:
Thanks for the data. I've added it to the repository and created a pull request to add it into libwacom.

I'm putting the finishing touches onto a pen ID patch for libwacom, after which GNOME will need to be updated to properly support them. As you notice, just adding a stylus definition isn't enough since GNOME doesn't know how to handle AES pen buttons right now.

closed 2019-10-14 by jigpu