linuxwacom/wacom-hid-descriptors #68: Sysinfo Lenovo Yoga 720

Trigtrig, 2019-10-23:
Hello everybody,

I am using a Lenovo Yoga 720 with the associated pen. The pen works well as far as I can tell. The pressure and the keys are recognized (tested with Xournal++ and Krita). Unfortunately I cannot configure the pen. In the gnome settings (Settings --> Devices --> Wacom Tablet --> Stylus) the pen is not displayed ("No stylus found"). Here is the result of the sysinfo script: 
[sysinfo.xiYkiceuGh.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/3762710/sysinfo.xiYkiceuGh.tar.gz)

Here is the result of xsetwacom:
`xsetwacom --list
Wacom HID 50EF Pen stylus       	id: 11	type: STYLUS    
Wacom HID 50EF Pen eraser       	id: 15	type: ERASER`

And this is the result of xinput list:
`xinput list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ MSFT0001:02 06CB:7F8F Touchpad          	id=10	[slave  pointer  (2)]
⎜   ↳ Wacom HID 50EF Pen stylus               	id=11	[slave  pointer  (2)]
⎜   ↳ Wacom HID 50EF Finger                   	id=12	[slave  pointer  (2)]
⎜   ↳ Wacom HID 50EF Pen eraser               	id=15	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Power Button                            	id=8	[slave  keyboard (3)]
    ↳ EasyCamera: EasyCamera                  	id=9	[slave  keyboard (3)]
    ↳ Ideapad extra buttons                   	id=13	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=14	[slave  keyboard (3)]`

Best regards,
Trig

jigpu, 2019-10-23:
Thanks for the sysinfo. I've added it to the repository and opened a pull request to add the a libwacom tablet definition. While that should allow the GNOME Control Center to recognize the sensor itself, in order for it to recognize the pen I will need to add a definition for your particular stylus.

In order to get the ID for your pen, can you please run our [capture.sh script](https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/af36c0a0fc2c019750472443369915ad7ea97737/capture.sh)? Once it is recording, just tap your stylus to the screen and then press CTRL+C to stop recording. Attach the generated tarball to this issue.

Trigtrig, 2019-10-25:
I executed the script. Here is the result: [record_1571989229.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/3770946/record_1571989229.tar.gz). If it's relevant, I used id 11 to record (see xinput list above). If you still need data, feel free to ask ;)

jigpu, 2019-10-28:
Thanks again :)

closed 2019-10-28 by jigpu