linuxwacom/wacom-hid-descriptors #70: Add support for HP Spectre x360 Convertible 15-df0xxx (2019)

ScorpionIT, 2019-11-03:
Stylus pointer works (two buttons no) but not recognized by kde system settings (kcm_libwacom)

ScorpionIT, 2019-11-06:
I have created elan-2817.tablet file based on other model

```
# ELAN touchscreen/pen sensor present in the HP Spectre x360 Convertible 15-df0xxx
[Device]
Name=ELAN 2817
ModelName=
DeviceMatch=i2c:04f3:2817
Class=ISDV4
Width=12
Height=7
IntegratedIn=Display;System

[Features]
Stylus=true
Touch=true
Buttons=0
```
and addedd this xorg config in /etc/X11/xorg.conf.d/ 

```
Section "InputClass"
    Identifier "Elan driver override"
    MatchUSBID "04f3:*"
    MatchDevicePath "/dev/input/event*"
    MatchIsTablet "true"
    Driver "wacom"
EndSection
```

now the tile pen is configurable and work very fine.

jigpu, 2019-11-19:
Thank you for the sysinfo output. Could you please open a pull request for https://github.com/linuxwacom/libwacom/ to add your tablet definition file?

closed 2019-11-19 by jigpu