linuxwacom/wacom-hid-descriptors #71: Add support for Lenovo Yoga C940 14IIL

Aurelien-, 2019-11-17:
The pencil works but is not detected in the settings. Battery level is not reported. The virtual keyboard does not show up when hitting a text field with the pencil. But if with the finger, virtual keyboard appears.

[sysinfo.MkSaCgsj6p.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/3855862/sysinfo.MkSaCgsj6p.tar.gz)

Thanks

jigpu, 2019-11-18:
Thanks for the data! I have added it to the repository and opened a pull request to add your sensor to libwacom.

Battery level should be reported through the GNOME Control Center's "Power" section when the pen is in use. It should be listed as an independent device. Do you see it there?

We are working on patches that should allow the pen to be configured in the GNOME Control Center in the future.

GNOME does not open the virtual keyboard when the pen is used, but if you would like to request that feature I believe the correct place would be https://gitlab.gnome.org/GNOME/gnome-shell/issues

closed 2019-11-18 by jigpu

Aurelien-, 2019-11-19:
Indeed, once the pen touches the screen the battery level appears! I will open a request feature for the virtual keyboard. Configuring the pen would be a must ! Thanks for your efforts.