linuxwacom/wacom-hid-descriptors #73: Dell Inspiron 13 7391

jclds139, 2019-12-06:
Under libinput, the pen and touch screen mostly work. Under xf86-input-wacom, nothing happens with either. The pen is Dell's cheaper PN350M, which is the same for [all current Inspiron models with digitizers.](https://www.dellemc.com/resources/en-us/asset/brochures/products/electronics-accessories/Dell_Pen_Compatibility_Brochure.pdf).

jclds139, 2019-12-06:
And I forgot the sysinfo file. Here it is: 
[sysinfo.M3bgVdHGzN.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/3930487/sysinfo.M3bgVdHGzN.tar.gz)

jigpu, 2019-12-09:
Thanks for sysinfo. I have pushed it up to the repository.

It looks like the pen and touchreen sensor is from Goodix. Such sensors are not normally picked up by xf86-input-wacom. You might investigate making a modification to `/usr/share/X11/xorg.conf.d/70-wacom.conf` as described in https://github.com/linuxwacom/xf86-input-wacom/issues/95 to force the driver to try and connect to it. If you're able to get it to work, you might also consider making a [libwacom tablet definition file](https://github.com/linuxwacom/libwacom/wiki/Tablet-Definition-Files) for the sensor.

closed 2019-12-09 by jigpu

jclds139, 2020-05-05:
Following those instructions to force the wacom driver, the touchscreen input is now detected and works fine, but the stylus no longer does anything, nor does it register in xinput at all. I suspect this isn't the place for this issue, but can you possibly point to where I should bring it up? If it helps, here's a new sysinfo dump in this state:
[sysinfo.d5zkeks3bQ.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/4583706/sysinfo.d5zkeks3bQ.tar.gz)

jigpu, 2020-05-13:
Its possible that the xf86-input-wacom driver simply doesn't handle the particular flavor of kernel events being sent by the Goodix pen. I can only suggest switching back to the libinput driver (possibly just for the pen, if that's what you want, by adding `MatchIsTouchscreen "false"` or `MatchIsTablet "true"` to your xorg.conf.d configuration). The libinput driver doesn't have as many pen-related features, but it _is_ the gold-standard for hardware compatibility... xf86-input-wacom will usually work with non-Wacom hardware, but we don't explicitly design it to.