linuxwacom/wacom-hid-descriptors #78: ThinkPad L13 Yoga

XrenynTheMage, 2020-01-18:
[sysinfo.wFXh0HNU4R.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/4080568/sysinfo.wFXh0HNU4R.tar.gz)

I've just bought a Lenovo ThinkPad L13 Yoga (this one: https://www.lenovo.com/de/de/laptops/thinkpad/l-series/ThinkPad-L13-Yoga/p/22TPL13L3Y1) and the touchscreen works like a charm. 
Its so-called "ThinkPad Pen Pro" however barely works, allowing me to point at things but not to draw.

for a more in-depth description of this problem, I'll refer over to my issue over at the libwacom project https://github.com/linuxwacom/libwacom/issues/198#issue-551013870

jigpu, 2020-01-24:
Thanks for the sysinfo. I've uploaded it to the repository and opened a pull request against libwacom.

closed 2020-01-24 by jigpu

talestomaz, 2020-05-04:
Hi @XrenynTheMage , I've just bought a L13 Yoga, installed Manjaro KDE e the pen is not working at all. Is it working fine for you? How did you make it? In Manjaro's forum people suggested me to have a look in the wacom-hid-descriptors here at GitHub, but I have no idea of programming and I'm really struggling to get what I'm supposed to do out of this. Could you help me?

jigpu, 2020-05-04:
@XrenynTheMage if the pen is not working at all (i.e., the pointer does not move in response to the pen), please open an issue at https://github.com/linuxwacom/input-wacom/issues/. If the pen works but does not show up in the control panel, you can file a new issue here. In either case, please attach the output from sysinfo.sh as described in the ["Contributing" section here](https://github.com/linuxwacom/wacom-hid-descriptors#contributing). This will give us some basic information about the hardware and state of the driver.

talestomaz, 2020-05-04:
Thanks @jigpu ! I've just opened an issue: https://github.com/linuxwacom/input-wacom/issues/177#issue-612088612. Yes, the problem is that there is no response to the pen. Nothing happens if I touch it on the screen... But sorry for the dummy question: how do I run such an script? Absolutely no idea...

talestomaz, 2020-05-04:
I think I figured out how to run the script. I copied the tex to Kate, saved it as .sh and run on terminal `sh script.sh`. I'm attaching the tar.gz file to the issue.