linuxwacom/wacom-hid-descriptors #81: Lenovo_ThinkPad_R61i_wacom_intous_s_under_UB18.04

WieslawSzafraniec, 2020-02-02:
My new wacom intous s (ctl-4100k) tablet is not recognized under UB 18.4.3. When I connect usb cable then cursor moves according to pen moves but Wacom settings (in device setup) was empty.
I check all .tablets files in /usr/share/libwacom/ directory and no one have usb id =056a:0374.
Can you help me?
[sysinfo.SnJcZT74MO.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/4143400/sysinfo.SnJcZT74MO.tar.gz)

jigpu, 2020-02-07:
Ubuntu 18.04 ships an out-of-date version of libwacom and a buggy version of the GNOME Control Center.

You can install an updated version of libwacom by following the [build instructions](https://github.com/linuxwacom/libwacom/wiki#building). This should allow you to configure some of the buttons on the tablet itself. Note that if you do this, be sure to check for the "duplicate match" issue afterwards by following the instructions [here](https://github.com/linuxwacom/libwacom/wiki#data-files).

If you need to configure both the pen and the tablet, you will need to update to at least Ubuntu 19.04 to get an updated version of libwacom and a less-buggy version of GNOME. You may still issues configuring all of the buttons on the tablet itself, but you should be able to also configure the pen.

closed 2020-02-07 by jigpu

WieslawSzafraniec, 2020-02-07:
Thank you for solution, not to easy ( for me :-)  )  but I can do it.
I used LTS release UB 18.04 and don't plane to go to 19.04. I thinking then LTS releases is under special attention and up -to-date .... :-(

jigpu, 2020-02-07:
Ubuntu 18.04 was likely up-to-date when it when it was released, but Ubuntu does not typically update our drivers afterwards. You may be able to contact them somehow and request an update to libwacom, but I have not had great luck in the past.