linuxwacom/wacom-hid-descriptors #97: Huawei Matebook HZ-W19

Ryuukyu313, 2020-07-22:
Hello, this is the gathered data from my Huawei Matebook HZ-W19 running a Wacom HID 4838.
[sysinfo.zIp2pgQHPA.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/4958860/sysinfo.zIp2pgQHPA.tar.gz)

jigpu, 2020-07-23:
Thanks! I've uploaded the sysinfo to the repository.

Before I submit a libwacom tablet definition, can you confirm that the pen and cursor stay in sync with each other? This device's HID descriptor looks odd: most sensors report a resolution of 2540 lpi, but yours only reports 254 lpi. If the pen and cursor stay in sync with each other, then its just a wierd quirk rather than a problem.

Ryuukyu313, 2020-07-23:
Thank you!

Yes, the pen stays in sync with the cursor just fine and works flawlessly.

jigpu, 2020-07-23:
Thanks for the confirmation. libwacom pull request has been opened.

closed 2020-07-23 by jigpu