linuxwacom/wacom-hid-descriptors #99: New Lenovo Laptop - touch pad does not work at all

jmbflat, 2020-09-06:
[sysinfo.81mrWrawDg.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5180358/sysinfo.81mrWrawDg.tar.gz)

Lenovo IdeaPad Flex 5 1511L05
Model 81x3

Blaimi, 2021-01-25:
I think you are facing the same probleme as described [here](https://askubuntu.com/q/1303753/238105). See also [this bug](https://bugzilla.kernel.org/show_bug.cgi?id=207759) on kernel.org. Especially comment 14 et. seq. provides a workaround solution.

jigpu, 2021-02-10:
The hardware in this device seem to match the other 15IIL05 that we have a record of, but its possible that Lenovo made some different variants. I see a WACF2200 sensor, however, which we've only recently been able to get working (see https://github.com/linuxwacom/input-wacom/issues/218). See https://github.com/linuxwacom/wacom-hid-descriptors/issues/139 for a description of how to install an updated kernel which should support the device.

jigpu, 2021-07-14:
@jmbflat are you able to submit updated sysinfo data? If you've updated to a more recent kernel release the hardware should work and the sysinfo script will finally be able to get useful information about it.

jmbflat, 2021-07-14:
@jigpu it looks like it is working with my last kernal update!  THanks for the follow-up

closed 2021-07-14 by jmbflat