linuxwacom/wacom-hid-descriptors #101: Support for Lenovo Yoga C640-13IML

PunkIslamist, 2020-10-02:
System info for Lenovo Yoga C640-13IML

[sysinfo.DhNRa5cW5x.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5318083/sysinfo.DhNRa5cW5x.tar.gz)

jigpu, 2020-11-04:
Can you reboot and then run `dmesg > dmesg.log` and attach the generated file here? It looks like the touchscreen could be a "WACF2200" device that was mentioned in sysfs but I can't find any other mention of the device or why it might not be working. Hopefully a full dmesg log could help with that.

richibrics, 2020-11-19:
Same problem for me

richibrics, 2020-11-19:
I have Lenovo Yoga C640-13IML as @PunkIslamist 
Here it is the hwinfo output that contains something inherent:

```

  P: /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7a/WACF2200:00
  L: 0
  E: DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7a/WACF2200:00
  E: SUBSYSTEM=acpi
  E: MODALIAS=acpi:WACF2200:PNP0C50:
  E: USEC_INITIALIZED=4345028
  E: ID_VENDOR_FROM_DATABASE=Wacom Tech
  

  P: /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-WACF2200:00
  L: 0
  E: DEVPATH=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-WACF2200:00
  E: SUBSYSTEM=i2c
  E: MODALIAS=i2c:WACF2200:00

/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7a/WACF2200:00

/devices/pci0000:00/0000:00:15.0
/devices/pci0000:00/0000:00:15.0/dma/dma0chan0
/devices/pci0000:00/0000:00:15.0/dma/dma0chan1
/devices/pci0000:00/0000:00:15.0/i2c_designware.0
/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1
/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-WACF2200:00
/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-dev/i2c-1
  name: /dev/i2c-1
/devices/pci0000:00/0000:00:15.0/idma64.0
```

I attach also the dmesg.log after reboot

[dmesg.log](https://github.com/linuxwacom/wacom-hid-descriptors/files/5566507/dmesg.log)

jigpu, 2020-11-20:
Thanks @richibrics. This problem seems to be a bit deeper than anticipated. I'm wondering if there's a problem with the ACPI table on these devices. Can you please run `sudo tar cvzf /tmp/acpi-tables.tar.gz /sys/firmware/acpi/tables/` and then attach the generated `/tmp/acpi-tables.tar.gz` file to this issue for review?

richibrics, 2020-11-20:
Sure
[acpi-tables.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5575741/acpi-tables.tar.gz)

jigpu, 2020-11-20:
Thanks, but it looks like some of the tables were corrupted. Reading up on this, it seems that we can't simply use `tar` to copy the files :frowning: I've written an alternate script for you to try; please download and save this script as [copy-acpi-tables.sh](https://gist.githubusercontent.com/jigpu/b16aed03c074524e9ff08f6276b959e3/raw/0edc8e49ed8363fb8dc6422c3866e780a4bebbc8/copy-acpi-tables.sh)

Run the script as `sudo ./copy-acpi-tables.sh` and then attach the `acpi-tables.tar.gz` tarball that it generates. Hopefully this version will be better.

richibrics, 2020-11-21:
Okay, else I could give you ACPI files from clover

richibrics, 2020-11-21:
Here you are, thanks for the support
[acpi-tables.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5577341/acpi-tables.tar.gz)

PunkIslamist, 2020-11-21:
Hello, sorry for my very late response :(
Is it helpful if I send you my files as well? Thankfully @richibrics has already provided his.

Slightly off-topic, but is the touchpad of the laptop working for you consistently @richibrics ? I kind of gave up with Linux on this model as the touchpad works for one or two boots, and then just stops working.

richibrics, 2020-11-21:
When I installed Arch on mine, it worked randomly (a boot yes and one not). Then I made it work (installed and edited a lot of things then I didn't know which one worked).
Two days ago I replaced Arch with Ubuntu and the error reappered, I tried to install (like I did on arch) the newest kernel version "5.9.0-050900" and it seems to be working fine now.

PunkIslamist, 2020-11-22:
> When I installed Arch on mine, it worked randomly (a boot yes and one not). Then I made it work (installed and edited a lot of things then I didn't know which one worked).
> Two days ago I replaced [...]

Those are good news! I will give it another shot then :)

jigpu, 2020-11-23:
Thanks for the updated copy of the tables. I've been reviewing them but haven't found anything conclusive. This particular Yoga does list an alternate device ID to be used if the installed OS isn't Win10-compatible. Can you try adding `acpi_osi="!Windows 2015"` to the end of the kernel command-line and then check if the pen works? If it still doesn't, attach the dmesg output so I can see if there's any difference with Win10-compatibility disabled.

See https://wiki.ubuntu.com/Kernel/KernelBootParameters for instructions on how to edit the kernel command line.

richibrics, 2020-11-23:
Same as before
[dmesg_2.log](https://github.com/linuxwacom/wacom-hid-descriptors/files/5584973/dmesg_2.log)

jigpu, 2020-11-30:
Thanks for giving it a shot. I'm a little surprised to not see it showing up with the new ID, but I'm admittedly new to deciphering ACPI tables so its entirely possible that this hack didn't have a chance of working anyway :(

FYI, I've opened a new centralized issue specifically for the WACF sensor detection issues at https://github.com/linuxwacom/input-wacom/issues/218. I'm going to close this specific issue for the time being.

closed 2020-11-30 by jigpu

jigpu, 2021-01-25:
Now that #218 has been fixed upstream, I would appreciate if @PunkIslamist or @richibrics could update to one of the required kernels and submit an updated sysinfo tarball. You should be able to see the pen show up in `xsetwacom list`, though I would be careful about actually trying to use the pen until #230 is fixed.

PunkIslamist, 2021-01-25:
> Now that #218 has been fixed upstream, I would appreciate if @PunkIslamist or @richibrics could update to one of the required kernels and submit an updated sysinfo tarball. You should be able to see [...]

First up, let me thank all the people that put in all the effort! That is truly amazing! I will try to test it during the weekend and upload the info. Will have to learn how to apply the updates to the kernel.

PunkIslamist, 2021-02-14:
> Now that #218 has been fixed upstream, I would appreciate if @PunkIslamist or @richibrics could update to one of the required kernels and submit an updated sysinfo tarball. You should be able to see [...]

Hello,
sorry for the late response once more. I hope it is still helpful, though.
[sysinfo.72igrHtWeC.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5977870/sysinfo.72igrHtWeC.tar.gz)

I installed the RC7 of kernel 5.11. I also played around a bit and it seems to work well. As already mentioned, using a pen kind of breaks it, but the pen itself seems to work well! I am using the Lenovo Pen Pro, in case that matters.

Once again, thanks for all the hard work!

jigpu, 2021-02-16:
@PunkIslamist thanks! I've pushed the data up to the repository and opened a pull request for libwacom to add the data file for your sensor.