linuxwacom/wacom-hid-descriptors #103: Support for Lenovo IdeaPad Flex 5 14IIL05

cfgnunes, 2020-10-11:
Touchscreen doesn't work on device:
Lenovo IdeaPad Flex 5 14IIL05

(also known as: IdeaPad Flex 5i)

I've tested on latest kernel 5.8 and many distros, such as Ubuntu 20.04.1.

[sysinfo.AN2KOXQI1D.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5362309/sysinfo.AN2KOXQI1D.tar.gz)

jigpu, 2020-11-04:
Can you reboot and then run dmesg > dmesg.log and attach the generated file here? It looks like the touchscreen could be a "WACF2200" device that was mentioned in sysfs but I can't find any other mention of the device or why it might not be working. Hopefully a full dmesg log could help with that.

jigpu, 2020-11-20:
This problem seems to be a bit deeper than anticipated. I'm wondering if there's a problem with the ACPI table on these devices. Can you please run `sudo tar cvzf /tmp/acpi-tables.tar.gz /sys/firmware/acpi/tables/` and then attach the generated `/tmp/acpi-tables.tar.gz` file to this issue for review?

ftsiadimos, 2020-11-23:
I have the same issue on Fedora 33.I am uploading the file in a case you need it.
[acpi-tables.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5586008/acpi-tables.tar.gz)

Machine:   Type: Convertible System: LENOVO product: 81X1 v: IdeaPad Flex 5 14IIL05 serial: <superuser/root required> 
           Mobo: LENOVO model: LNVNB161216 v: SDK0J40700 WIN serial: <superuser/root required> UEFI: LENOVO v: ECCN23WW

neonfighter28, 2020-11-24:
I have the same issue on arch 5.9.9-arch1-1. I am willing to contribute any information you might need :)

Alexfebbraio, 2020-11-25:
Hi, I have a lenovo flex 5 i3 10gen and I have the same problem with this ghost touckscreen wac2200 which is not detected by linux (I use ubuntu 20.10, fedora33 and chromeOS brunch)

neonfighter28, 2020-11-26:
I did some more digging and found some stuff that might help this issue. The device shows up in `/sys/bus/i2c/devices/i2c-WACF2200:00/` but the status in `./power/runtime_status` is `unsupported` . There is also another device there with the name `i2c-MSFT0001:00` and its `/power/runtime_status` is `unsupported` aswell. The weird thing is that the latter shows up in [xinput --list](https://gist.github.com/neonfighter28/e8fd1cd8761be4ffdd03bb6fc6cbe3ad) as my touchpad/mouse.

Alexfebbraio, 2020-11-26:
yes. 
I am following the Wacon thread, and have sent them my ACPI table as they requested

https://github.com/linuxwacom/input-wacom/issues/213

https://github.com/linuxwacom/wacom-hid-descriptors/issues/107

jigpu, 2020-11-30:
FYI, I've opened a new centralized issue specifically for the WACF sensor detection issues at https://github.com/linuxwacom/input-wacom/issues/218. I'm going to close this specific issue for the time being.

closed 2020-11-30 by jigpu

jigpu, 2021-01-25:
Now that #218 has been fixed upstream, I would appreciate if @cfgnunes or @neonfighter28 could update to one of the required kernels and submit an updated sysinfo tarball. You should be able to see the pen show up in `xsetwacom list`, though I would be careful about actually trying to use the pen until #230 is fixed.

neonfighter28, 2021-01-28:
Excuse the late response, here's the updated tarball running on self-compiled 5.10.2 with the patch applied. Pen works perfectly for me, and has never crashed anything

[sysinfo.n0ilTsFMOk.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5886010/sysinfo.n0ilTsFMOk.tar.gz)

jigpu, 2021-02-10:
Thanks! I've gone ahead and pushed the sysinfo up to the repository and opened a pull request to add support for this device to libwacom.