linuxwacom/wacom-hid-descriptors #107: Lenvo YOGA Duet 2020

liuqun, 2020-11-03:
OS: Ubuntu 20.10 and Ubuntu 20.04.1
Bug report:
1. The touchscreen does not work at all.
2. The touchpad works but the right button does not work.

[sysinfo.uPqAAX2Dpe.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5479561/sysinfo.uPqAAX2Dpe.tar.gz)

Lenovo store (China website):

https://item.lenovo.com.cn/product/1008173.html
[image]

jigpu, 2020-11-04:
Can you reboot and then run `dmesg > dmesg.log` and attach the generated file here? It looks like the touchscreen could be a "WACF2200" device that was mentioned in sysfs but I can't find any other mention of the device or why it might not be working. Hopefully a full dmesg log could help with that.

liuqun, 2020-11-05:
Here is the `sudo dmesg > dmesg.log` output:
[dmesg.log](https://github.com/linuxwacom/wacom-hid-descriptors/files/5491550/dmesg.log)
@jigpu dmesg

The device management program of Windows 10:
[image]
[image]

liuqun, 2020-11-05:
The following files are copied from the Winows 10 `C:\Windows\System32\drivers>` directory:
- WacHIDRouterISDU.sys
- mshidkmdf.sys

jigpu, 2020-11-05:
Thanks for the additional information. I've been reviewing everything and still can't quite figure out why the hardware isn't being detected properly. The only thing I've found of note so far is a small difference in sysfs:

**Working device (Lenovo Yoga C940-14IIL):** Note that the modalias value for both the `WCOM51E2:00` and `i2c-WCOM51E2:00` devices are `acpi:WCOM51E2:PNP0C50:`
~~~
********* /sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7d/WCOM51E2:00
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7d/WCOM51E2:00/uevent:MODALIAS=acpi:WCOM51E2:PNP0C50:
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7d/WCOM51E2:00/power_state:D0
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7d/WCOM51E2:00/power/runtime_active_time:0
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7d/WCOM51E2:00/power/runtime_active_kids:0
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7d/WCOM51E2:00/power/runtime_usage:0
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7d/WCOM51E2:00/power/runtime_status:unsupported
grep: /sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7d/WCOM51E2:00/power/autosuspend_delay_ms: Input/output error
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7d/WCOM51E2:00/power/async:disabled
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7d/WCOM51E2:00/power/runtime_suspended_time:0
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7d/WCOM51E2:00/power/runtime_enabled:disabled
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7d/WCOM51E2:00/power/control:auto
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7d/WCOM51E2:00/hid:WCOM51E2
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7d/WCOM51E2:00/path:\_SB_.PCI0.I2C0.TPL1
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7d/WCOM51E2:00/status:15
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7d/WCOM51E2:00/modalias:acpi:WCOM51E2:PNP0C50:

********* /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-WCOM51E2:00
/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-WCOM51E2:00/uevent:DRIVER=i2c_hid
/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-WCOM51E2:00/uevent:MODALIAS=acpi:WCOM51E2:PNP0C50:
/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-WCOM51E2:00/power/runtime_active_time:2990476
/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-WCOM51E2:00/power/runtime_active_kids:0
/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-WCOM51E2:00/power/runtime_usage:1
/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-WCOM51E2:00/power/runtime_status:active
grep: /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-WCOM51E2:00/power/autosuspend_delay_ms: Input/output error
/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-WCOM51E2:00/power/async:enabled
/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-WCOM51E2:00/power/runtime_suspended_time:1833
/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-WCOM51E2:00/power/runtime_enabled:enabled
/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-WCOM51E2:00/power/control:auto
~~~

**Non-working device (this system):** Note that the modalias values for `WACF2200:00` and `i2c-WACF2200:00` are different (`acpi:WACF2200:PNP0C50:` versus `i2c:WACF2200:00`). The missing `PNP0C50` might indicate a problem in the HID-over-I2C detection, and the change from `acpi` to `i2c` might indicate the wrong driver being assigned to the device.
~~~
********* /sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7c/WACF2200:00
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7c/WACF2200:00/uevent:MODALIAS=acpi:WACF2200:PNP0C50:
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7c/WACF2200:00/power/runtime_active_time:0
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7c/WACF2200:00/power/runtime_active_kids:0
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7c/WACF2200:00/power/runtime_usage:0
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7c/WACF2200:00/power/runtime_status:unsupported
grep: /sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7c/WACF2200:00/power/autosuspend_delay_ms: Input/output error
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7c/WACF2200:00/power/async:disabled
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7c/WACF2200:00/power/runtime_suspended_time:0
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7c/WACF2200:00/power/runtime_enabled:disabled
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7c/WACF2200:00/power/control:auto
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7c/WACF2200:00/hid:WACF2200
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7c/WACF2200:00/path:\_SB_.PCI0.I2C0.TPL1
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7c/WACF2200:00/status:15
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7c/WACF2200:00/modalias:acpi:WACF2200:PNP0C50:

********* /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WACF2200:00
/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WACF2200:00/uevent:MODALIAS=i2c:WACF2200:00
/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WACF2200:00/power/runtime_active_time:0
/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WACF2200:00/power/runtime_active_kids:0
/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WACF2200:00/power/runtime_usage:0
[... cut ...]
/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WACF2200:00/power/runtime_suspended_time:0
/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WACF2200:00/power/runtime_enabled:disabled
/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WACF2200:00/power/control:auto
/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WACF2200:00/name:WACF2200:00
/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WACF2200:00/modalias:i2c:WACF2200:00
~~~

jigpu, 2020-11-20:
Thanks @liuqun. This problem seems to be a bit deeper than anticipated. I'm wondering if there's a problem with the ACPI table on these devices. Can you please run `sudo tar cvzf /tmp/acpi-tables.tar.gz /sys/firmware/acpi/tables/` and then attach the generated `/tmp/acpi-tables.tar.gz` file to this issue for review?

liuqun, 2020-11-23:
[acpi-tables.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5581064/acpi-tables.tar.gz)

Here is the ACPI table. Thanks a lot for your reaction!

jigpu, 2020-11-23:
Thanks for the tables. I've been reviewing them but haven't yet found anything conclusive. The tables do list an alternate ID for the sensor, but I'm not sure if ACPI could be tricked into using it (and I don't know if that would even help, actually). I'll keep looking into this..

jigpu, 2020-11-30:
FYI, I've opened a new centralized issue specifically for the WACF sensor detection issues at https://github.com/linuxwacom/input-wacom/issues/218. I'm going to close this specific issue for the time being.

closed 2020-11-30 by jigpu

liuqun, 2021-01-05:
Bug fixed and tested with linux kernel 5.11.0-rc2

https://kernel.ubuntu.com/~kernel-ppa/mainline/v5.11-rc2/
- https://kernel.ubuntu.com/~kernel-ppa/mainline/v5.11-rc2/amd64/linux-headers-5.11.0-051100rc2-generic_5.11.0-051100rc2.202101032030_amd64.deb
- https://kernel.ubuntu.com/~kernel-ppa/mainline/v5.11-rc2/amd64/linux-headers-5.11.0-051100rc2_5.11.0-051100rc2.202101032030_all.deb
- https://kernel.ubuntu.com/~kernel-ppa/mainline/v5.11-rc2/amd64/linux-image-unsigned-5.11.0-051100rc2-generic_5.11.0-051100rc2.202101032030_amd64.deb
- https://kernel.ubuntu.com/~kernel-ppa/mainline/v5.11-rc2/amd64/linux-modules-5.11.0-051100rc2-generic_5.11.0-051100rc2.202101032030_amd64.deb

Fixed by Hui Wang from Ubuntu kernel team
See: Linux kernel patch at https://patchwork.kernel.org/project/linux-acpi/patch/20201211021814.36193-1-hui.wang@canonical.com/

jigpu, 2021-01-25:
Now that #218 has been fixed upstream, I would appreciate if @liuqun could update to one of the required kernels and submit an updated sysinfo tarball. You should be able to see the pen show up in `xsetwacom list`, though I would be careful about actually trying to use the pen until #230 is fixed.