linuxwacom/wacom-hid-descriptors #109: Lenovo Flex 14-API 81SS000BUS 1st Gen Active Pen and Touchpad stop working after an unknown amount of time.

TechieAndroid, 2020-11-19:
[sysinfo.MuVcg9kgJJ.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5564200/sysinfo.MuVcg9kgJJ.tar.gz)

TechieAndroid, 2020-11-19:
Let me know if you need any further information and I'll get that for you.

I'd love to contribute to this project and help out in any way I can, I'm unfamiliar with the process but maybe I could learn and be of some help.

jigpu, 2020-11-20:
Thanks for the sysinfo -- I've pushed it to the repository and created a pull request for libwacom to add the tablet definition file.

For the issue that you're seeing with the device stopping after some amount of time, please open an issue on the [input-wacom bug tracker](https://github.com/linuxwacom/input-wacom/issues/) so that we can figure out what's going on. You should add a link back to this issue so we have a quick reference to the sysinfo data.

As for contributing, is there anything in particular that interests you? We're stretched a bit thin at the moment, but I'd love to try to see what we can do :)

closed 2020-11-20 by jigpu

TechieAndroid, 2020-12-06:
Feel free to contact me on Discord or by email, Recompiler#9600 & vectorflaredesigns@gmail.com.
I've been learning C and Python, and studying kernel driver programming in C. I'm a newbie but I installed Gentoo Linux lately so I'm learning a lot, fast.