linuxwacom/wacom-hid-descriptors #111: MateBook HZ-W19 active stylus detected as finger

daxid, 2020-11-22:
Hi, 

On my MateBook HZ-W19, the active stylus is detected as my finger under Manjaro (confirmed using evtest).

However, it seems the events should be distinct :
```
/dev/input/event7:	Wacom HID 4834 Pen
/dev/input/event8:	Wacom HID 4834 Finger
```
(I also used to have an eraser reported, not sure why it does not show up right now...)

Could it be because the MateBook HZ-W19 was reported to run a Wacom 4838 but mine is 4834 ?

Best regards

daxid, 2020-11-22:
Here is my sysinfo : 

[sysinfo.WMalrfIR6D.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5580195/sysinfo.WMalrfIR6D.tar.gz)

daxid, 2020-11-22:
I can see the eraser with xsetwacom (but I'm usually underwayland)

```
 >>> xsetwacom list devices                                              [130]
Wacom HID 4834 Pen stylus       	id: 21	type: STYLUS    
Wacom HID 4834 Finger touch     	id: 22	type: TOUCH     
Wacom HID 4834 Pen eraser       	id: 29	type: ERASER
```

jigpu, 2020-12-02:
Thanks for the sysinfo, I've pushed it up to the repository and added a pull request for a new libwacom tablet definition file.

Please open a new issue at https://github.com/linuxwacom/input-wacom/issues/ for resolving the stylus/finger issue. Please run my [capture.sh script](https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/af36c0a0fc2c019750472443369915ad7ea97737/capture.sh) under X11, select the stylus device, and then draw a stroke or two with the pen. Press CTRL+C when done and then attach the captured tarball of logs to that issue so we can take a look at the data coming out of the hardware and drivers.

closed 2020-12-02 by jigpu