linuxwacom/wacom-hid-descriptors #112: Lenovo Idea Pad Flex 5 14ARE05

Weewoolad, 2020-11-25:
Partially working on Lubuntu with kernel 5.8.0-29. Attempted and failed with Void Linux with kernel 5.8. Attempted and failed with Lubuntu with kernel 5.10.

[acpi-tables.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5599151/acpi-tables.tar.gz)
[sysinfo.fGWhYaLpPe.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5599154/sysinfo.fGWhYaLpPe.tar.gz)

jigpu, 2020-11-30:
Thanks for the sysinfo. I've added it to the repository. Its also great news that the device is partially working in Lubuntu -- I'll be taking an extra-close look at this sysinfo in particular :)

FYI, I've opened a new centralized issue specifically for the WACF sensor detection issues at https://github.com/linuxwacom/input-wacom/issues/218. I'm going to close this specific issue for the time being.

closed 2020-11-30 by jigpu

jigpu, 2021-01-25:
Now that #218 has been fixed upstream, I would appreciate if @Weewoolad could update to one of the required kernels and submit an updated sysinfo tarball. You should be able to see the pen show up in `xsetwacom list`, though I would be careful about actually trying to use the pen until #230 is fixed.