linuxwacom/wacom-hid-descriptors #123: Dell XPS 13 9370

bashelier, 2021-01-06:
it seems the Wacom One Pen is not well recognized on a Ubuntu 18.04 ruuning on a Dell XPS 13 9370, as reported by dmesg :

`[    3.091190] wacom 0003:056A:03A6.0004: Unknown device_type for 'Wacom Co.,Ltd. Wacom One Pen Display 13'. Ignoring.`

Then, the gnome settings does not show the Pen, but shows the tablet.

Please note, the tablet and the pen were working well with the default distribution packages but didn't appear at all in the gnome settings.
The tablet appeared in the gnome settings after manual installatiion of the libwacom from github.

After manual installation of input-wacom kernel module there is still no way to see the pen appearing in gnome settings.
Then the only way to configure the Pen is by using xsetwacom command, but it is volatile.

May be the informations attached in this issue will allow you to fix that.
[sysinfo.cPmHPtItIr.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5776660/sysinfo.cPmHPtItIr.tar.gz)

jigpu, 2021-01-06:
This specific "Unknown device_type" error is normal. The driver is complaining about one of the additional USB interfaces which is not necessary for proper function. You can safely ignore it in this case.

The most likely reason for your control panel issues is an out-of-date libwacom database. The control panel will only allow you to configure tablets which are defined in this database, but Ubuntu 18.04 predates the release of the Wacom One. The simplest solution is to copy the tablet definition from the current libwacom release onto your system and then reboot:

1. Download https://raw.githubusercontent.com/linuxwacom/libwacom/master/data/wacom-one.tablet
2. Move the file to /usr/share/libwacom
3. Reboot
4. Ensure the tablet is connected
5. Check the control panel to see if the tablet now shows up

>  gnome settings does not show the Pen, but shows the tablet.

Can you clarify what you mean? I assume you mean that you see a "Wacom Tablet Settings" (or similar) control panel option but nothing appears inside of it?

bashelier, 2021-01-06:
Thank you for your answer.

Indeed, I already had an up-to-date version of the wacom-one.tablet file. Probably thanks to the manual installation of the libwacom.

I mean the "Wacom Tablet" panel shows 2 buttons : one for displaying the Stylus settings and one for the Tablet settings.
It is ok for the tablet settings, which allows to calibrate it.
But the stylus settings page shows nothing else but a message asking to move the stylus to the proximity of the tablet in order to be able to configure it (as mentioned here : https://help.gnome.org/users/gnome-help/stable/wacom-stylus.html.en). 
I failed to find the way to display the stylus settings menu, whatever close to the tablet/screen the stylus is, or any action I try...
However it should be possible to configure it a bit, at least to assign a specific action/key to the button.

I am not sure it is an issue to this project or a gnome bug. If you have any clue...

jigpu, 2021-01-06:
Thanks, that clarifies the situation quite a bit :) GNOME has historically had some problems detecting the kinds of pen used by the Wacom One. The current GNOME release works correctly, but its possible that the version you have installed does not. Can you check on the GNOME version for me? If you open the "About" section of the control panel you should find that it lists the GNOME version. I can then try to see if that version would be expected to have the fix or not.

bashelier, 2021-01-07:
Well I am sorry to be quite confuse ;-)
My GNOME version is 3.28.2.
Many thanks for your help !

jigpu, 2021-01-11:
I believe this issue was fixed by [this commit](https://gitlab.gnome.org/GNOME/gnome-control-center/-/commit/7c85d5a871b2996a55cae83d995991800bcefb6d) which was integrated into GNOME 3.36. Since that version of GNOME should be in Ubuntu 20.04 you might want to try creating a live disk with it to see if it recognizes the stylus for your Wacom One.

bashelier, 2021-02-24:
Hi. Sorry for the delay.
You're right. It was fixed in the gnome version of Ubuntu 20.04.
Thank you.

closed 2021-02-24 by bashelier