linuxwacom/wacom-hid-descriptors #124: Huion H950P drawing tablet

jconnolly837, 2021-01-09:
[66.41.13.62:170/download/sysinfo.FklgWpPH14.tar.gz](66.41.13.62:170/download/sysinfo.FklgWpPH14.tar.gz)
Sorry, I couldn't figure out how to attach files on github. I hope a self hosted thing is ok.

jigpu, 2021-01-11:
I'm having trouble getting the download link to work, unfortunately. To attach files here on Github you can just drag-and-drop them onto the text box you write your reply into. You can also click on the banner just below the text box that says how to attach files.

jconnolly837, 2021-01-11:
[sysinfo.FklgWpPH14.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5796460/sysinfo.FklgWpPH14.tar.gz)
Ok, now it's working.

jigpu, 2021-01-11:
It looks like your tablet's Vendor ID was not in the list checked by the script, so only partial logs were returned. I've just updated the script with a few more vendor IDs and it should work better now. Please [re-download the sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and then run/upload the revised logs.

jconnolly837, 2021-01-11:
[sysinfo.5rJ4iuOykV.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5796565/sysinfo.5rJ4iuOykV.tar.gz)

jigpu, 2021-01-11:
Thanks again! Logs look good, so I've pushed them up to the repo. Go ahead and add a link to this issue in your libwacom tablet definition :)

(libwacom issue for future reference: https://github.com/linuxwacom/libwacom/pull/308)

closed 2021-01-11 by jigpu