linuxwacom/wacom-hid-descriptors #133: Huion HS610

filoumoteur, 2021-01-24:
Here's the script ran with a tablet HUION HS610 attached to the computer.

[sysinfo.HUION_HS610.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5862755/sysinfo.HUION_HS610.tar.gz)

jigpu, 2021-01-25:
Thanks! I've pushed this info up to the repository. It looks like libwacom already has this device in its database so there shouldn't be anything else do do.

closed 2021-01-25 by jigpu

filoumoteur, 2021-01-25:
Hello!
To get `libwacom-list-local-devices` to correctly identify my device, I manually created a .tablet file with the configuration of the tablet. Without it, I wasn't able to configure it and it wasn't identified by `xsetwacom list`.
Perhaps I should remove my config file before generating the sysinfo?

jigpu, 2021-01-25:
No, that's not necessary. If your tablet definition was manually created, I would suggest opening an issue at https://github.com/linuxwacom/libwacom/issues requesting they integrate your tablet definition. Include a link to this issue as well as a copy of your tablet definition file. They should be able to review the file and let you know if any changes are necessary and then help you get it integrated into libwacom for the benefit of other users.