linuxwacom/wacom-hid-descriptors #137: HP Pro Tablet 408

ShapeShifter499, 2021-02-04:
[sysinfo.B81FfSnxIO.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5926957/sysinfo.B81FfSnxIO.tar.gz)

ShapeShifter499, 2021-02-04:
I reran the sysinfo script as root since I noticed some stuff like dmesg logs were blank
[sysinfo.4e4IRHrhyF.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5927003/sysinfo.4e4IRHrhyF.tar.gz)

ShapeShifter499, 2021-02-04:
One more taken under xorg mode gnome and the proper xorg utilities installed, the other two dumps were taken in wayland gnome mode. 

[sysinfo.hyVgebcvky.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5927138/sysinfo.hyVgebcvky.tar.gz)

jigpu, 2021-02-04:
It looks like this uses a sensor from Synaptics, which the sysinfo script currently ignores (most of their stuff is just touchpads). Please download the following patch file and move it to the same location as the sysinfo script. Next, open a terminal, navigate to the directory with both the sysinfo script and patch, and run `patch -p2 < sysinfo.patch.txt` to add Synaptics support. Finally re-run the sysinfo script and attach its output.

[sysinfo.patch.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/5927141/sysinfo.patch.txt)

ShapeShifter499, 2021-02-04:
here you go
[sysinfo.lYkzorH9EG.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5927573/sysinfo.lYkzorH9EG.tar.gz)

What does this mean for adding proper support. While the stylus seems to work fine, I would like to see it show up properly in Gnome settings and other programs that make use of similar information.

ShapeShifter499, 2021-02-04:
I have one other question about this. I saw in the generated xinput log that there is a "eraser" like I saw on another tablet I own, a Surface Go.  Now I know the two have different stylus technology. But it makes me wonder if there might be a Synaptics stylus out there somewhere that has a convent eraser tip on the other end of the pen like the Surface Pens.  

@jigpu Do you by any chance know if such a thing exists?

jigpu, 2021-02-05:
Thanks, that did the trick alright. I've pushed the sysinfo up to the repository for later reference.

The first step to getting the device showing up in the control panel is to create a libwacom tablet definition for it. The sysinfo script would normally take care of this for a Wacom device, but since this is Synaptics you'll need to write one from scratch. You can find a basic introduction at the [Tablet Definition Files wiki page](https://github.com/linuxwacom/libwacom/wiki/Tablet-Definition-Files) and a more complete template in the [wacom.example file](https://github.com/linuxwacom/libwacom/blob/master/data/wacom.example). You can also check out the [other existing definitions](https://github.com/linuxwacom/libwacom/tree/master/data) (I'd recommend using the "isdv4" files for reference since they're all built-in sensors like this). You can test your definition locally by saving it to the `/usr/share/libwacom` directory and running `libwacom-list-local-devices`. When you're happy with the result, either [file a libwacom issue](https://github.com/linuxwacom/libwacom/issues) with your tablet definition attached or create a pull request.

As for the second question, the sensor's HID descriptor is capable of reporting eraser status but I don't know if any such stylus actually exists. Its possible that one does, but its also possible that no such pen was ever made, or that the sensor doesn't _actually_ support erasers despite claiming otherwise (it wouldn't be the first time...). Its also possible that the eraser could be set up like many of Wacom's AES pens, where a button on the side of the stylus activates the eraser rather than a second tip on the back of the pen.

closed 2021-02-05 by jigpu

ShapeShifter499, 2021-02-06:
>(I'd recommend using the "isdv4" files for reference since they're all built-in sensors like this)

@jigpu is there is a reason why I wouldn't want to mark it as a "Class=PenDisplay" instead of "Class=isdv4"?

jigpu, 2021-02-09:
The `PenDisplay` class is used by simple peripheral display tablets like Wacom's various business-oriented products, so wouldn't really be appropriate here. Conversely, the `ISDV4` class is used for all kinds of system-integrated sensors, even from other manufacturers.

It isn't super critical to get the class correct since it seems to be mostly unused by applications, but we still try to follow existing precedent just in case.

ShapeShifter499, 2021-02-11:
@jigpu I noticed there are a database of stylus IDs, should I get the ID of the stylus used for the particular stylus used for this tablet? 

How would I even do that if I could?

This is the stylus information from what I can see on the inside of the AAAA battery compartment.
HP Pro Tablet 408 Active Pen
HP Product Number: 797838-001
Model: PR77S (HP)
SPS No.: 798051-001
S/N: 7CF6050HCG

jigpu, 2021-02-11:
I'm not sure if your particular stylus supports tool IDs, but to find out you'll need to run `sudo libinput record > active-pen.txt`, select the pen device (I believe it will be named something like `SYNA7508`), and then drag the pen across the screen to generate some events. Afterwards, press CTRL+C to stop recording and attach the `active-pen.txt` file here for review.

The Wacom kernel driver sends the tool ID (if supported) as an `ABS_MISC` input event. We can check to see if your pen sends any events of that type.

ShapeShifter499, 2021-02-11:
I had two identically named events on my system so I recorded from both. I swiped the stylus across the screen and clicked all of the buttons while it was in range of the screen. 

```/dev/input/event1 SYNA7508:00 06CB:125D``` > [active-pen_event1.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/5968538/active-pen_event1.txt)

```/dev/input/event2 SYNA7508:00 06CB:125D``` > [active-pen_event2.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/5968543/active-pen_event2.txt)

Recording from event2 gave me this message at the end ```No events recorded, deleting 'stdout'``` which I'm guessing has nothing to do with anything stylus related. 

I don't notice any stylus IDs in the output but correct me if I'm wrong.  In any case as it stands right now (Febuary 2021) I think this tablet is as well supported by the software as it can be. Even pressure sensitivity seems to work fine in applications like Xournal++

jigpu, 2021-02-11:
Yup, looks like its just a basic sensor / stylus. Leaving out the stylus information should be fine :+1:

ShapeShifter499, 2021-02-11:
Perfect!  Thank you for all the help.