linuxwacom/wacom-hid-descriptors #140: HP ENVY x360 Convertible 15-dr1144ng

Thomi7, 2021-02-09:
It looks like that this kind of device has been added before in [https://github.com/linuxwacom/wacom-hid-descriptors/issues/102](https://github.com/linuxwacom/wacom-hid-descriptors/issues/102), but I noticed that wayland has been used to gather the logs. Additionally there has been no table file for that device in that time.

Thomi7, 2021-02-09:
[sysinfo.piFZRjWWDX.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5950145/sysinfo.piFZRjWWDX.tar.gz)

Here is a new log file gather in xorg and with the new table file: [https://github.com/linuxwacom/libwacom/pull/347](https://github.com/linuxwacom/libwacom/pull/347)

jigpu, 2021-02-09:
Thanks! Sysinfo has been pushed to the repository.

closed 2021-02-09 by jigpu