linuxwacom/wacom-hid-descriptors #142: Touch works but starts touching from a distance

hashim19, 2021-02-11:
Hi, 
 
My system details are attached. I have a PN771M pen which came with the dell inspiron 7306 laptop. Touch works partially, however, when I take the pen closer to the screen it starts its actions even before I make any contact. Moreover, I can not find where can I configure my stylus pen. 

[sysinfo.q6kjnRuwa7.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/5963461/sysinfo.q6kjnRuwa7.tar.gz)

jigpu, 2021-02-11:
Thanks; I've pushed this info up to the repository. Stylus configuration may only possible under a GNOME Wayland session, and only with an appropriate libwacom tablet definition. You'll want to file an issue at https://github.com/linuxwacom/libwacom/issues and reference this bug report so they can quickly find this sysinfo. You can try to piece together a tablet definition for them based on [the example](https://github.com/linuxwacom/libwacom/blob/master/data/wacom.example) or request that they use this sysinfo to generate one themselves.

Unfortunately, there's nothing we can do about the pen sensitivity. Its possible that the pen tip is damaged (e.g. from being dropped) and now always reports a non-zero pressure.

closed 2021-02-11 by jigpu

Swivelgames, 2021-05-24:
I'm not sure if it helps to add additional info, but I went ahead and generated a sysinfo of my machine as well.

[sysinfo.b5fNksXqwQ.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/6535466/sysinfo.b5fNksXqwQ.tar.gz)

Interestingly, I'm able to get the pen working (it's a Dell PN771M, and the Digitizer and Pen are manufactured by Goodix Shenhzen), but it's always selecting text, even when the pen is not touching the screen. It appeared to work with most of the existing classes, with some of them refusing to click at all (I believe the Touchscreen classes), and others constantly registering Button 1. The latter was achieved with the "Wacom N-Trig class".

The pen appears to have two, or maybe three buttons. A button on the eraser-side of the pen, a button near the index finger, and the tip of the pen is able to compress, which may actuate another button as well (though I'm not sure).

Swivelgames, 2021-05-25:
Additional info can be found here, but I'm not sure how far this avenue is going to get: https://gitlab.com/AdyaAdya/goodix-touchscreen-linux-driver/-/issues/20