linuxwacom/wacom-hid-descriptors #157: Thinkvision M14t

Jaydon-chai, 2021-06-07:
Thanks for the work you guys do. It's much appreciated.

[sysinfo.1ni27Z3QGq.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/6605999/sysinfo.1ni27Z3QGq.tar.gz)

Jaydon-chai, 2021-06-07:
oh yeah, it works pretty well even without being detected, although touch layer is sometimes mapped to inbuilt display rather than to the monitor. 
my guess is that wayland has some heuristic that doesn't work amazing since my laptop display and M14t monitor have the same resolution.
https://www.reddit.com/r/kde/comments/cub3by/touchscreen_order_wayland/

Jaydon-chai, 2021-06-07:
Here's the tar generated when the script was run under sudo, I was checking through the files to make sure everything was in order and xrandr logs and xinput don't work since Im running fedora 34, GNOME 40.1 with wayland, so no X11. Otherwise, sudo gives a bit more.

[sysinfo.4trjkKChOo.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/6606053/sysinfo.4trjkKChOo.tar.gz)

jigpu, 2021-06-21:
Do you know whether this device was set to run in "Windows" or "Android/Chrome" mode when you created this log? I'd be interested in having logs for both modes so that we know how the device appears in both circumstances.

Jaydon-chai, 2021-06-26:
It was in Windows mode when I created this log, I'll do another one now for Android/Chrome

Jaydon-chai, 2021-06-26:
Here's the one for android/chrome
[sysinfo.xBFmtL95fP.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/6719634/sysinfo.xBFmtL95fP.tar.gz)

Jaydon-chai, 2021-06-26:
Here's the one for windows again just in case
[sysinfo.RDyYyBWgLx.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/6719635/sysinfo.RDyYyBWgLx.tar.gz)

jigpu, 2021-07-13:
Apologies for the delay. I've uploaded the data for both modes to the repository and opened a pull request for libwacom. It looks like the hardware is exposed identically in both Windows and Android / Chrome modes, so the difference in behavior is down to how data gets sent over its devices.

closed 2021-07-13 by jigpu