linuxwacom/wacom-hid-descriptors #169: Huion H1161

jorgesca, 2021-09-08:
Hello, my Huion H1161 graphic tablet is detected as the model H950P. The tablet sort of works, but this model has more buttons (10) than the H950P. It also has a touch strip (which from what I found so far is not supported in gnome).
I've been reading about configuring HUION tablets but honestly I feel a bit lost.
The one issue that made me realize my tablet was not being detected correctly, which I am not sure is related to libwacom at all, is that in some apps I can't erase with the pen (specifically in pdfpc, a pdf present
er console which allows me to draw over PDF slides).  
I run the sysinfo.sh script from gnome on wayland. I can run it from Xorg if required.

[sysinfo.cOBStqOaDx.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/7128463/sysinfo.cOBStqOaDx.tar.gz)

edoiks, 2021-12-07:
I have the same issue? Is there something that might help more? any additional logs?

edoiks, 2021-12-08:
My file attached, I tried to generate .tablet file by myself, but tablet does not work.
[sysinfo.Rh9aAsD3Ag.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/7672507/sysinfo.Rh9aAsD3Ag.tar.gz)

jigpu, 2022-01-07:
Thanks for the sysinfo -- it has been added to the repository. You'll need to file a [libwacom issue](https://github.com/linuxwacom/libwacom/issues) with a link back here to have the tablet definition updated. It looks like the H640P and H950P use the same VID:PID as this H1161...

closed 2022-01-07 by jigpu