linuxwacom/wacom-hid-descriptors #170: Lenovo Flex 5 14ALC05

zerotao, 2021-09-15:
Another varient:

[sysinfo.YPpNpkeE1A.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/7173867/sysinfo.YPpNpkeE1A.tar.gz)

zerotao, 2021-09-15:
git-update does not generate a .tablet (at least not for me)

zerotao, 2021-09-15:
looks like the device match is now: i2c:056a:5218

zerotao, 2021-09-16:
Not certain about how correct this one is (particularly the width/height) but this one now works isdv4-5218.tablet:

```# Lenovo IdeaPad Flex 5 14ALC05
# Sensor Type: AES
# Features: Touch (Integrated), Tilt
# HW Resolution: 30931 x 17399 (2540 x 2540 lpi)
#
# Autogenerated from sysinfo.LyIiLqPZux
# https://github.com/linuxwacom/wacom-hid-descriptors/issues/164#issue-941998063

[Device]
Name=ISDv4 5218
ModelName=
DeviceMatch=i2c:056a:5218
Class=ISDV4
Width=12
Height=7
IntegratedIn=Display;System
Styli=@isdv4-aes;

[Features]
Stylus=true
Touch=true
Buttons=0```

jigpu, 2022-01-07:
Thanks for the sysinfo -- it has been added to the repository. An auto-generated tablet definition has also been submitted to libwacom.

The size data for your tablet definition matches up with the HID descriptors so should be fully functional :)

closed 2022-01-07 by jigpu