linuxwacom/wacom-hid-descriptors #177: HP Spectre x360 Convertible 14

HonbraDev, 2021-11-11:
Doesn't show up in GNOME Settings in Ubuntu 21.10 and Fedora 34

Tried adding `i2c:04f3:2beb` to DeviceMatch in `/usr/share/libwacom/elan-2514.tablet`. The "tablet" now gets recognized, but the stylus still doesn't.

The touchscreen, pressure sensitivity and buttons work.

When I bring the stylus close to the screen, a notification pops up telling me its battery is at 1% (even after charging).

[image]

[image]

[image]

[sysinfo.S0QOERWHbT.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/7523025/sysinfo.S0QOERWHbT.tar.gz)

HonbraDev, 2021-11-11:
After rebooting into Wayland, the stylus now gets recognized, however, the button mapping doesn't work.

HP has a bundled Windows utility for modifying the buttons, but I couldn't get it to run under WINE and I don't have Windows installed at the moment. I'll try to upload the utility here, as I still have a backup.

HonbraDev, 2021-11-11:
The "backup" I was referencing 10 minutes ago was an AnonFiles link that is no longer alive.

Why AnonFiles? I wanted to minimize Windows' (and the bloatware HP preinstalled) exposure to the Internet.

Lesson learned: don't use AnonFiles.

jigpu, 2022-01-07:
Thanks for the sysinfo -- it has been added to the repository. If you haven't already submitted your tablet definition to libwacom, you can file an issue [here](https://github.com/linuxwacom/libwacom/issues) for its inclusion in future releases. Be sure to include a link here for reference to the sysinfo.

As for the button mapping not working, the GNOME desktop has some known issues that sometimes prevent it from working. See e.g. https://gitlab.gnome.org/GNOME/gnome-control-center/-/issues/1020

closed 2022-01-07 by jigpu