linuxwacom/wacom-hid-descriptors #178: Lenovo Thinkpad 11e Yoga gen 6

emhl, 2021-11-24:
the sysinfo for my laptop
[sysinfo.J40s9ticHs.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/7595425/sysinfo.J40s9ticHs.tar.gz)

jigpu, 2022-01-07:
Thanks for the sysinfo -- it has been added to the repository. The info could not be automatically converted to a tablet definition, so you'll want to file an issue at https://github.com/linuxwacom/libwacom/issues with a link back here for additional help.

closed 2022-01-07 by jigpu

pjanx, 2024-02-23:
Perhaps like this, if anyone needs to fix autorotate in GNOME.
```
# ELAN touchscreen/pen sensor present in the Lenovo Thinkpad 11e Yoga gen 6
[Device]
Name=ELAN 901C
ModelName=
DeviceMatch=i2c:04f3:2a40
Class=ISDV4
Width=12
Height=7
IntegratedIn=Display;System
Styli=@generic-no-eraser;

[Features]
Stylus=true
Touch=true
```