linuxwacom/wacom-hid-descriptors #181: HP ZBook x2 G4 Detachable Workstation

himekifee, 2021-12-17:
The machine is powered by `wacom emr` technology. Both stylus and touch screen can be utilised but only stylus shows in `xinput`. Not able to configure the touch in `kde wacom` configuration tab.
[sysinfo.52jLcHT93O.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/7731257/sysinfo.52jLcHT93O.tar.gz)

jigpu, 2022-01-07:
Thanks for the sysinfo -- its been pushed to the repository. For some reason the sysinfo script was not able to detect the touchscreen. This is a bit odd since it can detect most manufacturers (Wacom, Elan, Synaptics, Microsoft) but its possible your device uses something more custom. Would you mind sharing the full output from the `dmesg` command after a fresh boot? That may give us a little more to go on...

Since the touchscreen is not a Wacom device, it is probably being powered by the libinput drivers. If KDE has a control panel for other touchscreens you might have better luck there.

himekifee, 2022-01-10:
Sorry for the delay. Forgot to mention the touch screen device is listed as `ATML1000:00 03EB:8ABB`. It is categorised as a slave pointer in `xinput` but can be properly detected as a touch screen in `xournalpp`.

jigpu, 2022-01-11:
The sysinfo script didn't have Atmel in its list of vendors to scan, so that would explain why it didn't find anything. If you can download and run the [updated version](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh), it should not contain some actual sensor info. Attach the new tarball here and I'll push its data to the repository.

Afterwards, you might need to file a bug on the [libinput bug tracker](https://gitlab.freedesktop.org/libinput/libinput/-/issues) if the touchscreen does not work as you expect. Does it work on the desktop (i.e. dragging windows around) at all?

himekifee, 2022-01-11:
[sysinfo.IZU4bPJ05N.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/7845399/sysinfo.IZU4bPJ05N.tar.gz)
It kinda works including touch and dragging windows. Palm rejection is not working though, I feel the difference between this one and another tablet which has the `wacom` device as a USB device. That one works flawlessly.

jigpu, 2022-01-25:
I've updated the database to include the new sysinfo with the Atmel info.

Getting palm rejection to work properly may be tricky since the pen and touch input are being handled by different drivers that have no idea about the state of each other. You may have some luck by modifying your config files to force the two devices to use the same driver (i.e. move the Atmel touchscreen to xf86-input-wacom or the Wacom pen to xf86-input-libinput). Try saving the following snippet as `/etc/X11/xorg.conf.d/90-zbook-pentouch.conf` and rebooting:

~~~
# Force HP ZBook x2 G4 touchscreen and pen devices to use the same driver.
# Only *one* of the blocks below should be uncommented!

Section "InputClass"
        # Force the Atmel touchscreen to use the Wacom driver
        Identifier "HP ZBook x2 G4 touchscreen OVERRIDE"
        MatchUSBID "03EB:8ABB"
        MatchDevicePath "/dev/input/event*"
        Driver "wacom"
EndSection

#Section "InputClass"
#        # Force the Wacom pen to use the libinput driver
#        Identifier "HP ZBook x2 G4 stylus OVERRIDE"
#        MatchUSBID "056A:016C"
#        MatchDevicePath "/dev/input/event*"
#        Driver "libinput"
#EndSection
~~~

Leaving the first block uncommented will cause both both devices to use the xf86-input-libinput driver (which should provide the best pen experience, but may not be ideal for the touchscreen). Modifying it so only the second block is uncommented will cause both devices to use the xf86-input-libinput driver (which may be more ideal for the touchscreen but less for the pen).

closed 2022-01-25 by jigpu

himekifee, 2022-01-25:
I'm using the first block since `KDE` did not provide a config interface for `libinput wacom` device. Also, it doesn't seem to be a palm rejection in `libinput` for tablet with touchscreen. For the block, I used `MatchIsTouchscreen` instead of `MatchUSBID` since the touchscreen is on the i2c bus so not USB at all. After that, everything seems to be working fine including palm rejection.

himekifee, 2022-01-25:
In that case, would it be helpful to toggle `Touch=false` to `true`?

jigpu, 2022-01-26:
> In that case, would it be helpful to toggle `Touch=false` to `true`?

The `Touch=false` setting shouldn't have much of an effect on how the pen/touch work. It's primarily intended to let control panels know whether they should show touch-related options for the device (which isn't necessary for the EMR device).